rust-embedded/cortex-m #9: should we add `std::sync` like primitives based on critical sections?

japaric, 2016-11-01:
See the [mutex] branch for an implementation of a mutex based on critical sections. And [this PR] for an example that uses that mutex to share state between the main thread and an interrupt.

The advantage of implementing them using critical sections is smaller code size, faster locking and that they don't require the LDREX or STREX instructions which are not available in the Cortex-M0(+).

If you have and can afford atomics then you should probably use the [parking_lot] crate.

[mutex]: https://github.com/japaric/cortex-m/commits/mutex
[this PR]: https://github.com/japaric/f3/pull/58
[parking_lot]: https://crates.io/crates/parking_lot

cc @thejpster

thejpster, 2016-11-01:
I like this approach, but I think I'd prefer to follow the std::sync::Mutex API, or at least offer that as well.

```
let mut data = data.lock().unwrap();
*data += 1;
```

Could that work?

japaric, 2016-11-01:
I prefer the more explicit scopes that closures provide and don't really like the release on drop mechanism because the guard can be "forgotten" using `mem::forget` but I'll sketch something and get back to you.

japaric, 2017-03-12:
There's a `Mutex` in tree.

closed 2017-03-12 by japaric