rust-embedded/cortex-m #11: Critical section token

awelkie, 2016-11-19:
What do you think about having a struct like zinc's [NoInterrupts](https://github.com/hackndev/zinc/blob/master/src/hal/cortex_common/irq.rs#L23) struct and passing it into the closure for the `interrupt::free` function? The idea is to make a zero-sized struct that can't be created by users and is only ever created in a critical section. Then, if a function should only be called in a critical section it can take a reference to that struct which will guarantee that it will only be called in a critical section.

It would look something like this
```
mod interrupts {
    pub fn disable() {}
    pub fn enable() {}

    pub struct CritToken {
        _private: ()
    }

    pub fn free<F>(f: F)
        where F: FnOnce(&CritToken)
    {
        disable();

        f(&CritToken{ _private: () });

        enable();
    }
}

fn dangerous_func(tok: &interrupts::CritToken) {}

fn calling_dangerous_func() {
    interrupts::free(|tok| {
        dangerous_func(tok);
    });
}
```

japaric, 2016-11-19:
Sounds like a good idea to me. As long as the token doesn't control *when* the interrupts are disabled/enabled but, instead, it's only an indicator that the interrupts are disabled. i.e. `CritToken` doesn't implement `Drop` like zinc's `NoInterrupts` does.

japaric, 2017-03-12:
done in #17

closed 2017-03-12 by japaric