rust-embedded/cortex-m #24: consider enabling interrupts in reset handler

whitequark, 2017-05-06:
It's something of a footgun: since the body of main must be protected by `cortex_m::interrupt::free` whether I want it or not, it's easy to assume that interrupts are enabled outside of that critical section, and forget to add an explicit `cortex_m::interrupt::enable();` after it.

It does seem odd when coming from C, but with interrupt safety by default, I can't think of any real downsides...

japaric, 2017-05-07:
Not sure I follow. Interrupt *are* re-enabled once `interrupt::free` ends. Or are you referring to enabling specific exceptions / interrupts using Nvic.enable?

whitequark, 2017-05-07:
I mean interrupts are disabled when initially entering `main`, but I would like to see them enabled when initially entering `main`. I want that because it is more convenient and just as safe.

pftbest, 2017-05-08:
And since `interrupt::enable()` is now `unsafe`, you would always have `unsafe` code in your `main()`

japaric, 2017-05-08:
OK ... I must be missing something.

- Interrupt are neither enabled or disabled (PRIMASK is not touched at all) during the reset handler before `main` is called.
- If you use `interrupt::free` you temporarily disable interrupts (PRIMASK = 1) and then reenable them when that ends.

So unless there are some microcontrollers that boot with all interrupts disabled (PRIMASK = 1) -- the stm32 micros certainly don't do this -- then I don't see why you would need to call `interrupt::enable` in `main` at all. Interrupts already Just Work without that.

whitequark, 2017-05-08:
> So unless there are some microcontrollers that boot with all interrupts disabled (PRIMASK = 1)

I believe this is the case with TM4C. I will double-check.

whitequark, 2017-05-08:
I must have made a mistake somewhere. TM4C does not boot with PRIMASK=0.

closed 2017-05-08 by whitequark

pftbest, 2017-05-08:
I always thought that interrupts are disabled in main() and manually enabled them. Don't know why I did it this this way.