rust-embedded/cortex-m #32: Doesn't work with `CARGO_INCREMENTAL`.

dpc, 2017-05-15:
It took me a long while to debug why linker script behaves erratically and sometimes compiles right, but other times complaints either about the lack of exception handlers or too many of them if I attempt to add them. Turns out, all it took is to disable `CARGO_INCREMENTAL` environment variable.

japaric, 2017-05-15:
You should report this in rust-lang/rust if it's not there already. AFAIR there was an issue open about incremental compilation having problems with lang items / linker_section and resulting in linker errors. You should check if that's the same error you have been running into, and if not you should open a new issue.

japaric, 2017-05-16:
Added a warning about this to the how to use and the troubleshooting sections of the cortex-m-quickstart [documentation](https://docs.rs/cortex-m-quickstart/0.1.5/cortex_m_quickstart/#used-cargo_incremental1).

japaric, 2017-05-16:
Oh, and I *think* the cause the problem is that incr. comp. is ignoring `#[linker_section]` directives but I haven't had a chance to investigate that just yet.

closed 2017-05-16 by japaric

dpc, 2017-05-16:
I think that would fit my symptoms, yes.