rust-embedded/cortex-m #45: stdout for the exception handler

a-shafir, 2017-06-12:
Currently if and exception or a non-implemented interrupt triggered the _default_handler_ function called. In case if no debugger connected it appears as freeze. This can be way more friendly by adding output to std-out as the configurable feature:
`extern "C" fn handler(_sr: &StackedRegisters) -> ! {`
`#[cfg(exception_print = "stdout")]`
`{`
`    println!("EXCEPTION {:?} @ PC=0x{:08x} LR=0x{:08x} XPSR=0x{:08x}", Exception::current(), _sr.pc, _sr.lr,
        _sr.xpsr);`
`    println!("R0=0x{:08x} R1=0x{:08x} R2=0x{:08x} R3=0x{:08x} R12=0x{:08x}", _sr.r0, _sr.r1, _sr.r2, _sr.r3, 
        _sr.r12);`
`}`
or similar.

japaric, 2017-06-15:
I'm planning a revamp of the interrupt / exception registration mechanism that will make it possible to override the `default_handler` function. Then users will be free to do whatever makes sense for their application in that function.

> This can be way more friendly by adding output to std-out as the configurable feature:

But there's no such thing as an *standard* output in embedded. Some applications may want to use serial, others semihosting and yet others the ITM. And I certainly don't want to provide a feature for all the possible options. The `defaut_handler` override will handle this case without adding a bunch of Cargo features.

a-shafir, 2017-06-23:
What about using panic! as the default _standard_ output? Panic already has the override.

japaric, 2017-06-24:
panic! is not a good choice for stdout because it will terminate (\*) the program when invoked. I don't think it's appropriate in this case either exactly because panic! behavior is up to the application. panic! could not print anything at all, it could trigger a bkpt, it could call intrinsics::abort causing another hard fault which would probably result in infinite recursion.

(\*) Unless you have an unwinding mechanism. But it any case panic! signals "something bad happened".

a-shafir, 2017-06-24:
The current exception.rs triggers bkpt and unconditional infinity loop. When no debugger connected there is no way to see what happens in any case. If a panic! used in many (most) of the cases it will print. Of course use a dedicated separate stdout driver and stack for the exception/panic but it also can be done in the overrided panic! function.

jonas-schievink, 2019-11-28:
Panic implementations can now be chosen and implemented freely by the user, and both the hard fault as well as the default interrupt handler can be overridden using cortex-m-rt. Since there is no standard output on embedded, these will have to suffice.

(It would be cool to have retargetable stdio streams in libcore, but that's an issue that needs to go through Rust's RFC process)

closed 2019-11-28 by jonas-schievink