rust-embedded/cortex-m #58: Privileged mode

kjetilkjeka, 2017-07-11:
There is currently no support for working in privileged mode.

There should at least be
- a way to call the svc instruction
    - This can be implemented by ```cortex_m::asm::svc(num: u32)```
- a way to write thread mode privilege level
    - This can be implemented by ```cortex_m::register::control::write(Control)``` and perhaps also something in the lines of ```cortex_m::register::control::modify(f: F) where f: FnOnce(Control) -> Control```

The cortex_m_rt crate seem to already implement the possibility for overriding the SVC handler. So with these changed it should be possible to: 
- Execute the work in need of privileged mode from the svc handler
- Set the thread mode from the SVC handler and then execute the work after returning from the handler.

kanishkarj, 2019-08-14:
What is the status of this? Currently, I am having to write assembly code and create pendSV interrupt manually. I figured that out from [here](https://github.com/n-k/cortexm-threads/blob/master/src/lib.rs#L258). Is there any better way ?

korken89, 2019-08-14:
It is this you are referring to? https://docs.rs/cortex-m/0.6.0/cortex_m/peripheral/struct.SCB.html#method.set_pendsv
This will pend the SVCall exception.

korken89, 2019-08-14:
The true SVC instruction is a bit problematic, as this requires `asm!("svc 123" :::: "volatile")`, which is not allowed on stable and the `123` is not allowed to be a register which thwarts the current workaround of the `asm.s` file.

However, if you run the nightly compiler it should be no issue, you can use the functions here for reference https://github.com/rust-embedded/cortex-m/blob/master/src/asm.rs

Or if you want stable, create your own `my_asm.s` is which you could do something like this:
```
  .section .text.__svc_123
  .global __svc_123
  .thumb_func
__svc_123:
  svc 123
  bx lr
```

kanishkarj, 2019-08-15:
Oh yeah, I did come across those nightly compiler features. I guess I have to work with stable, so the second option I guess. Thank you :)

kanishkarj, 2019-08-31:
Hey, also is there any way to Run code in privileged mode. Also is there anyway to make your code use the Main stack pointer instead of the process stack pointer.

jonas-schievink, 2019-09-15:
@kanishkarj Privileged mode and Main Stack Pointer (MSP) are used by default after reset of Cortex-M processors. The issue title and description are incorrect, I'm assuming it's talking about entering **un**privileged mode (and using the Process Stack Pointer instead of MSP).

kanishkarj, 2019-09-15:
Yeah, I realised. I don't have so much experience with embedded programming. I got to know that MSP is used by default recently. Well yes, I would like to know how to change to unprivileged mode ? I assume it is possible to do it with the PR mentioned below.

jonas-schievink, 2020-01-14:
It's now possible (as of 0.6.2) to switch to unprivileged mode by writing to the `CONTROL` register with the `npriv` field set to `Npriv::Unprivileged`. Switching stack pointers is not possible using this API, since the necessary write to the SPSEL bit is only possible in thread mode, but at the same time will lose the current stack, which is unsafe. I believe this fundamentally needs a short asm sequence to do something useful, and what that is is fundamentally application-dependent, so cortex-m shouldn't try to provide more APIs there.

One more nice-to-have for dealing with unprivileged mode is a way to use the `svc` instruction. As outlined in https://github.com/rust-embedded/cortex-m/issues/58#issuecomment-521264947 this is rather difficult to do without inline asm, and this also wouldn't allow doing many *useful* things like invoking syscalls (since no parameters can be passed with this approach). This makes such a wrapper only useful in very few circumstances in which `PendSV` or another exception works just as well.

I suggest closing this issue, since the APIs the cortex-m provides are pretty much the best it can provide already.

jonas-schievink, 2020-08-29:
Since there were no objections, closing as per the comment above.

closed 2020-08-29 by jonas-schievink