rust-embedded/cortex-m #74: Should ITM write functions check if a debugger is connected like CMSIS ITM_SendChar ?

jothan, 2018-01-12:
I am currently unable to run my application standalone when a debugger is not present. I am suspecting that it is because the TPIU and the ITM port are not enabled by my code.

Would it be a good idea to check if tracing is enabled like ITM_SendChar ?

https://github.com/ARM-software/CMSIS/blob/6891719834ac6ca2a79e8ded00620faffae10ae8/CMSIS/Include/core_cm3.h#L1698

If so, I would like to submit a pull request to short circuit write_all if the debugger is not enabled.

japaric, 2018-03-10:
> Would it be a good idea to check if tracing is enabled like ITM_SendChar ?

I would prefer not to slow down the ITM logging functionality as I use it log data at high data rates so low overhead is a must.

> I am suspecting that it is because the TPIU and the ITM port are not enabled by my code

There's ITM initialization code in #82 but it requires unsafe code. That issue is about making the initialization code safe; I would push for that instead of this issue.

nickray, 2019-03-16:
Just ran into this issue too, agree that the write functions should not do a non-zero cost check. But the documentation might want to mention that they never return if the ITM is not enabled as in https://github.com/rust-embedded/cortex-m/issues/82.

jonas-hagen, 2019-06-22:
I experienced this issue as well, took me a while to find out whats going wrong. Because it only happens if the debugger is not connected, it is a bit hard to debug.

Since ITM can be enabled and **disabled** by the debugger at any time, even if we have proper ITM enable methods, this would still be an issue.

What about having the check done in the `iwrite` macros just to keep beginners (like me) away from this issue?

bcantrill, 2020-06-09:
Just came across this issue; assuming this issue is being seen on ARMv8-M parts (vs. ARMv7-M), it seems quite likely that it is in fact a duplicate of #201.