rust-embedded/cortex-m #79: Peripheral definitions in cortex-m are not `Send`, while those from `svd2rust` generated crates are

jmesmon, 2018-01-17:
Comparing something like:
 - [`stm32f103xx::TIM6`](https://docs.rs/stm32f103xx/0.8.0/stm32f103xx/struct.TIM6.html)
 - [`cortex_m::ITM`](https://docs.rs/cortex-m/0.4.1/cortex_m/peripheral/struct.ITM.html)

Notably:
 - `Send` is omitted from `cortex_m::ITM`
 - mutable access (in the form of `ptr() -> *mut RegisterBlock` and `DerefMut` are removed from `stm32f103xx::TIM6`

I ran into this while trying to pass around `ITM` in `cortex_m_rtfm`, which wants `Send` on `resources`:

```
error[E0277]: the trait bound `*const (): core::marker::Send` is not satisfied in `cortex_m::peripheral::ITM`
  --> src/main.rs:16:1
   |
16 | / app! {
17 | |     device: stm32f429,
18 | |
19 | |     resources: {
...  |
29 | |     },
30 | | }
   | |_^ `*const ()` cannot be sent between threads safely
   |
   = help: within `cortex_m::peripheral::ITM`, the trait `core::marker::Send` is not implemented for `*const ()`
   = note: required because it appears within the type `core::marker::PhantomData<*const ()>`
   = note: required because it appears within the type `cortex_m::peripheral::ITM`
```

I've used a quick wrapper to workaround:
```
...
use core::ops::{Deref,DerefMut};
pub struct ITM(cortex_m::peripheral::ITM);

impl Deref for ITM {
    type Target = cortex_m::peripheral::ITM;
    fn deref(&self) -> &Self::Target {
        &self.0
    }
}

impl DerefMut for ITM {
    fn deref_mut(&mut self) -> &mut Self::Target {
        &mut self.0
    }
}
unsafe impl Send for ITM {}

...
app! {
...
    resources: {
        static R_ITM: ITM;
        ...
    },

    tasks: {
        TIM7: {
            path: periodic,
            resources: [R_ITM, ...],
        },
    },
}

...
fn init(mut p: init::Peripherals) -> init::LateResources {
    ...

    init::LateResources {
        R_ITM: ITM(p.core.ITM),
        R_TIM7: tim,
    }
}

...
```

japaric, 2018-01-17:
> Send is omitted from cortex_m::ITM

Whoops. All the peripherals should implement `Send` in this crate; that was my omission. I'll cook up a PR.

closed 2018-01-17 by homunkulus

japaric, 2018-01-17:
v0.4.2 which contains the fix for this has been published

commit c921d43f (2018-01-17): Auto merge of #80 - japaric:send, r=japaric

impl Send for all the peripherals

closes #79
PR #80 (2018-01-17): impl Send for all the peripherals
closes #79