rust-embedded/cortex-m #87: Fails to build:  can't find crate for `compiler_builtins`

et1975, 2018-04-12:
On OSX:

`rustc -V`
>rustc 1.27.0-nightly (ad610bed8 2018-04-11)

`rustup target add thumbv7m-none-eabi`
>info: downloading component 'rust-std' for 'thumbv7m-none-eabi'
info: installing component 'rust-std' for 'thumbv7m-none-eabi'

` cargo build --target thumbv7em-none-eabi`
>   Compiling vcell v0.1.0
   Compiling bare-metal v0.1.1
   Compiling aligned v0.1.1
   Compiling untagged-option v0.1.1
   Compiling cortex-m v0.4.3 (file:///[snip]/cortex-m)
error[E0463]: can't find crate for `compiler_builtins`
error[E0463]: can't find crate for `compiler_builtins`
error[E0463]: can't find crate for `compiler_builtins`
error[E0463]: can't find crate for `compiler_builtins`

>error: aborting due to previous error

>For more information about this error, try `rustc --explain E0463`.
error: aborting due to previous error
error: aborting due to previous error
error: aborting due to previous error

>For more information about this error, try `rustc --explain E0463`.
For more information about this error, try `rustc --explain E0463`.
For more information about this error, try `rustc --explain E0463`.
error: Could not compile `aligned`.
warning: build failed, waiting for other jobs to finish...
error: Could not compile `untagged-option`.
warning: build failed, waiting for other jobs to finish...
error: Could not compile `vcell`.
warning: build failed, waiting for other jobs to finish...
error: Could not compile `bare-metal`.

What am doing wrong?

japaric, 2018-04-12:
> rustup target add thumbv7m-none-eabi

> cargo build --target thumbv7em-none-eabi

The target names don't match. You have to install the rust-std component for thumbv7em-none-eabi, i.e. `rustup target add thumbv7em-none-eabi`.

et1975, 2018-04-12:
That was it, thank you!

closed 2018-04-12 by et1975