rust-embedded/cortex-m #92: SCB missing FPSCR, FPCAR, FPCCR

dlrobertson, 2018-05-31:
I see the CPACR in `scb`, but I do not see the Floating-point Status and Control Register (FPSCR), Floating-point Context Address Register (FPCAR), or Floating-point Context Control Register (FPCCR). AFAIK this is only an optional feature with Cortex-M4 and Cortex-M7 and should be included along with the CPACR, but I could be mistaken. See the [User Guide](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0553a/BABGHFIB.html) for more info.

japaric, 2018-08-03:
Hmm, you'll find FPSCR, FPCAR, FPCCR behind the FPU proxy. Now, why is CPACR behind the SCB proxy and not the FPU proxy ... I'm not sure. It seems more consistent to have CPACR in FPU as well but that would be a breaking change.

japaric, 2018-08-07:
> As a result, the FPU struct would need a lot of reserved fields right?

That's not an issue. As long as one register doesn't show up in two register blocks / proxies we are fine.

@rust-embedded/cortex-m this needs a decision.

korken89, 2018-08-07:
It would indeed be more consistent to have it with the `FPU`, I generally value consistency high and would argue to change in any case.
Comments from the rest of @rust-embedded/cortex-m ?

adamgreig, 2018-08-07:
The ARMv7-M architecture reference lists CPACR with the rest of the SCB registers while the FP ones are 'additional SCB registers for the FP extension'; in theory you could have a cortex-m device with _other_ coprocessors which were enabled using CPACR that were not the ARM FPU (not that I know of any such devices).

[image]

So logically I'm not sure it makes sense for CPACR to live in the FPU struct. It enables the FPU but only in the same sense that an RCC peripheral enables clocks to other peripherals; it isn't itself part of the FPU.

adamgreig, 2019-03-12:
I propose closing this issue and leaving the registers as-is, with FPSCR, FPCAR, and FPCCR in the FPU peripheral and CPACR in the SCB. Any thoughts?

dlrobertson, 2019-03-12:
Agreed

closed 2019-03-12 by adamgreig