rust-embedded/cortex-m #95: Unable to run cargo doc over project.

fud, 2018-06-21:
I am using this crate indirectly from the stm32f1 crate when I try to document that library I'm getting an error with cortex-m.

cargo doc -p stm32f1 --open

` Documenting cortex-m v0.5.2
error: `[::task::Poll]` cannot be resolved, ignoring it...
   |
note: lint level defined here
  --> /Users/billsb/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.2/src/lib.rs:42:9
   |
42 | #![deny(warnings)]
   |         ^^^^^^^^
   = note: #[deny(intra_doc_link_resolution_failure)] implied by #[deny(warnings)]
   = note: the link appears in this line:

            [`Poll::Pending`](::task::Poll). The future will *also* register the
                              ^^^^^^^^^^^^
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: `[::task::Context::waker]` cannot be resolved, ignoring it...
   |
   = note: the link appears in this line:

            [`cx.waker()`](::task::Context::waker)). Once a task has been woken up,
                           ^^^^^^^^^^^^^^^^^^^^^^
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: Could not document `cortex-m`.`

japaric, 2018-06-21:
Thanks for the report.

>         [`Poll::Pending`](::task::Poll). The future will *also* register the

That doc comment doesn't exist in this crate. It actually comes from the `core` crate (src/libcore/future.rs). What you are experiencing is a bug in rustdoc; it actually looks like a mixture of rust-lang/rust#51468 and rust-lang/rust#51661.

I don't think there's much we can do in this crate to fix this problem though. You could try generating the documentation using an older nightly release using something like `cargo +nightly-2018-06-01 doc -p stm32f1 --open` (note: haven't tried this myself; you may need to use a different date)

fud, 2018-06-21:
Thanks for your detailed reply @japaric I will have a try of that tonight.

closed 2018-06-21 by fud

MajorBreakfast, 2018-07-07:
Should be fixed with https://github.com/rust-lang/rust/pull/51853

fud, 2018-07-10:
I can confirm that this fix did indeed work. Cheers.