rust-embedded/cortex-m #96: Errors in documentation?

korken89, 2018-06-25:
Hi, when running `cargo doc`, the following errors are produced for this package:

```
 Documenting cortex-m v0.5.2
error: `[::task::Poll]` cannot be resolved, ignoring it...
   |
note: lint level defined here
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.2/src/lib.rs:42:9
   |
42 | #![deny(warnings)]
   |         ^^^^^^^^
   = note: #[deny(intra_doc_link_resolution_failure)] implied by #[deny(warnings)]
   = note: the link appears in this line:
           
            [`Poll::Pending`](::task::Poll). The future will *also* register the
                              ^^^^^^^^^^^^
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: `[::task::Context::waker]` cannot be resolved, ignoring it...
   |
   = note: the link appears in this line:
           
            [`cx.waker()`](::task::Context::waker)). Once a task has been woken up,
                           ^^^^^^^^^^^^^^^^^^^^^^
   = help: to escape `[` and `]` characters, just add '\' before them like `\[` or `\]`

error: Could not document `cortex-m`.
```

I hope this helps to fix it :)

chrysn, 2018-06-28:
The bad reference is part of the core documentation comments (and only hits cortex-m because it denies warnings), and is still there in 2018-06-27 nightly (but was not there in stable 1.27, which works fine).

I think that the breakage is related to `task::Poll` being behind the futures_api feature gate (but the Future trait that returns a Poll and mentions it in its docs is behind the same gate).

Until the root cause is found, I suggest that the `#![deny(warnings)]` be overridden by an `#[allow(intra_doc_link_resolution_failure)]`.

japaric, 2018-08-03:
This appears to be fixed as of nightly-2018-08-01.

closed 2018-08-03 by japaric