rust-embedded/cortex-m #101: when stable, use `core::arch::arm` to implement `cortex_m::asm`

japaric, 2018-08-02:
this will let us provide the `cortex_m::asm` API on stable without depending on an external assembler.

japaric, 2018-08-27:
Removing from the RC1 (edition) milestone because thanks to PR #108 this crate no longer on GCC so this is just a nice to have.

jonas-schievink, 2020-08-29:
We now use Rust's new inline assembly (https://github.com/rust-lang/rust/issues/72016) and an FFI shim to implement `cortex_m::asm`, removing the need for any external tooling. Inline asm has higher chances of being stabilized than `core::arch::arm`, so maybe we should close this issue? (doesn't seem too important to keep tracking this; we can remove the FFI shim once inline asm is stable)

adamgreig, 2020-08-31:
SGTM. I agree that inline asm getting stabilised looks more likely than all the relevant instructions landing in core::arch::arm, and in any event we need a few assembly routines for some methods anyway.

closed 2020-08-31 by adamgreig