rust-embedded/cortex-m #113: `SHCSR` vs `SHCRS` naming issue?

jamesmunns, 2018-09-04:
Currently, in the SCB peripheral, there is [a field named `shcrs`]. However, [the ARM manual] refers to this register (the "System Handler Control and State Register") as `SHCSR`.

[the ARM manual]: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/Cihieffb.html
[a field named `shcrs`]: https://github.com/rust-embedded/cortex-m/blob/ee0ff685a0bdf903144cbb9db1881f63c21b8992/src/peripheral/scb.rs#L50-L51

I can open up a PR to change, however I wanted to check I am not missing something first :)

japaric, 2018-09-04:
There's an open PR for that, #97, but it's a breaking change so it won't land until we are ready to make a new minor release.

jamesmunns, 2018-09-04:
Whoops, thanks @japaric!

closed 2018-09-04 by jamesmunns