rust-embedded/cortex-m #125: cortex_m::asm::delay() doesn't seem to work on thumbv6m

therealprof, 2018-12-07:
While attempting to create an example for https://github.com/rust-embedded/wg/issues/259 to blink an LED using `cortex_m::asm::delay` I'm greeted with this interesting error:
```
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt;
extern crate panic_halt;

extern crate stm32f042_hal as hal;

use hal::prelude::*;
use hal::stm32;

use cortex_m::asm::delay;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    let p = stm32::Peripherals::take().unwrap();
    let gpioa = p.GPIOA.split();
    let mut led = gpioa.pa1.into_push_pull_output();

    loop {
        led.set_high();
        delay(8_000_000);

        led.set_low();
        delay(8_000_000);
    }
}
```

```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "/Users/egger/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv6m-none-eabi/lib" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/examples/blinky-b437c53f5b36f500.blinky.60cthedq-cgu.1.rcgu.o" "-o" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/examples/blinky-b437c53f5b36f500" "--gc-sections" "-L" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/deps" "-L" "/Users/egger/OSS/stm32f042-hal/target/release/deps" "-L" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/build/cortex-m-563ed60a5fde6333/out" "-L" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/build/cortex-m-rt-414ccd4438657f1e/out" "-L" "/Users/egger/OSS/stm32f042-hal/target/thumbv6m-none-eabi/release/build/stm32f0-f3baf49fdf34c49d/out" "-L" "/Users/egger/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv6m-none-eabi/lib" "--start-group" "-Bstatic" "/var/folders/8f/ch8vbtq93ldfc4vksdp5cw540000gn/T/rustcjqym5j/libcortex_m_rt-f3ac26fff634eea2.rlib" "/var/folders/8f/ch8vbtq93ldfc4vksdp5cw540000gn/T/rustcjqym5j/libcortex_m-945aa0e87bd515a7.rlib" "--end-group" "/Users/egger/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv6m-none-eabi/lib/libcompiler_builtins-962f36023deecec7.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: /var/folders/8f/ch8vbtq93ldfc4vksdp5cw540000gn/T/rustcjqym5j/libcortex_m-945aa0e87bd515a7.rlib(cortex-m.o):(.text.__delay+0x4): unrecognized reloc 103
```

simonsso, 2018-12-16:
I found this bugreport when I was investigating a similar bug building for a nordic nrf51 (Cortex-M0)
enable feature flag inline-asm let me pass it. The question is if this should be default for m0 somehow?

[dependencies.cortex-m]
version = "0.5.8"
default-features = false
features = ["inline-asm"]

sajattack, 2019-01-03:
Anyone have a clue as to the cause? I just ran into this today. The inline-asm feature defeats the point because then you can't use stable.

japaric, 2019-02-08:
I can see this when using rust-lld to link the binary but not when using `arm-none-eabi-ld`. According to this [overflow answer](https://stackoverflow.com/a/47725251) the problem seems to be that the relocation *type* used by the arm-none-eabi-as used to assemble the binary blob we ship with the crate is not known by the lld that's shipped with rustc.

These are two potential fixes: wait until lld learns about this new relocation type, or build the binary blobs with an older version of `arm-none-eabi-as`.

japaric, 2019-02-08:
> build the binary blobs with an older version of `arm-none-eabi-as`.

I tested this idea with `arm-none-eabi-as` 2.24 (Ubuntu 14.04) and it didn't work; I still get the relocation error.

closed 2019-07-04 by bors[bot]