rust-embedded/cortex-m #128: Move the singelton macro to the bare-metal crate?

andre-richter, 2019-01-09:
Hi all,

IMO the singleton macro is nothing `cortex-m` specific, it would also apply to other architectures and CPUs. It would make sense moving it to https://github.com/japaric/bare-metal. We have done something like that in the past already I see (#50).

Thoughts?

BR,
Andre

japaric, 2019-01-09:
It is Cortex-M specific in its current form though; it contains [a call to interrupt::free](https://github.com/rust-embedded/cortex-m/blob/db6718650505d09fc9a959fe0ffa4b41d2fc1d4b/src/macros.rs#L55) which is Cortex-M specific and defined in this crate.

If you want to move `singleton!` to the bare-metal crate you would need to change the API to something like this:

``` rust
use bare_metal::singleton;
use cortex_m::interrupt; // or cortex_r::interrupt or msp430::interrupt or etc.

let buf = interrupt::free(|cs| {
    // cs has type CriticalSection which is defined in the bare-metal crate
    singleton!(cs, : [u8; 128] = [0; 128]);
});
```

Alternatively, you can implement `singleton!` using atomics instead of a critical section:

``` rust
fn main() {
    // this
    let buf = singleton!(: [u8; 128] = [0; 128]);

    // expands into
    let buf: Option<&'static mut [u8; 128]> = {
        static TAKEN: AtomicBool = AtomicBool::new(false);

        if !TAKEN.compare_and_swap(false, true, Ordering::AcqRel) {
            static mut X: [u8; 128] = [0; 128];

            Some(unsafe { &mut X })
        } else {
            None
        }
    };
}
```

But this won't work on Cortex-M0, MSP430 and probably other architectures. It will work on Linux systems (like x86_64-Linux) though.

andre-richter, 2019-01-09:
Ah, I overlooked that the interrupt::free line is already baked into it. Sorry.
My question was motivated by wanting to reuse the singleton pattern for cortex-a based code, but here we have atomics (at least for aarch64, not sure about earlier archs).

Adding your atomics-based proposal to the `cortex-a` crate might in fact be the better solution.

closed 2019-01-09 by andre-richter