rust-embedded/cortex-m #131: New release?

adamgreig, 2019-01-28:
We might be due another release. A patch release would land #130 which is blocking thumbv8 in cortex-m-rt and other repos, while a minor release could also land outstanding #97, #126, and #127 (and perhaps we could do #106 as well).

Any thoughts @rust-embedded/cortex-m ?

therealprof, 2019-01-28:
Big fan of RERO.

korken89, 2019-01-29:
Sounds good to me

korken89, 2019-02-05:
Ping @rust-embedded/cortex-m

adamgreig, 2019-02-05:
Any thoughts on minor vs patch? I'm happy to prep for either, though I'm not sure what precisely we'd have to do for the semver trick.

thejpster, 2019-02-15:
This is blocking something I'm working on. Anything I can do to push it out the door?

(I also need https://github.com/rust-embedded/cortex-m-rt/pull/167, which this blocks)

thejpster, 2019-02-15:
Let's get #97 in and go to 0.6.0. No point holding back.

korken89, 2019-02-16:
SGTM

Disasm, 2019-02-22:
Do we have any consensus on this? There are 3 PRs waiting for the minor release.

thejpster, 2019-02-22:
https://twitter.com/42Technology/status/1099009843967471617

I can't publish the nRF9160 demo until we get this in, as I've had to vendor and patch the whole `cortex-m.*` set of crates.

korken89, 2019-02-23:
Ping @rust-embedded/cortex-m - if there are no objections withing 12 hours from now, I will merge the PRs and do the 0.6.0 release.

japaric, 2019-02-23:
We have a process for things like this. Using voting and the accepted reduced majority RFC you could have gotten a minor release approved in like 2 weeks, 2 weeks ago. In any case, this seems to be the tally for a new minor release:

- [ ] @adamgreig 
- [x] @korken89 
- [x] @thejpster 
- [ ] @therealprof 
- [ ] @ithinuel 
- [ ] ~@japaric~ abstains

I will abstain from participating this time as I want to see the team handle a release without me.

> I'm not sure what precisely we'd have to do for the semver trick.

The semver trick is documented [here](https://github.com/dtolnay/semver-trick). Consider what happens if you mix code that depends on 0.5.x and code that depends on 0.6.0, whether the semver trick can be applied in this case or not and whether applying the semver trick actually helps or not.

Go team!

therealprof, 2019-02-26:
@korken89 We do have a voting majority. Still interested in cutting the release?

korken89, 2019-02-26:
I can do the release, however I was thinking about what @japaric was talking about:

> Consider what happens if you mix code that depends on 0.5.x and code that depends on 0.6.0, whether the semver trick can be applied in this case or not and whether applying the semver trick actually helps or not.

Are there any ideas here, as linking in 2 cortex-m crates might cause issues with symbol names (I am on a conference right now and can't give it a test). Has anyone given it a try?
I can test it more tomorrow when I am back home, and try to mitigate if there are issues.

thejpster, 2019-03-12:
Any more thoughts on this anyone?

therealprof, 2019-03-12:
@thejpster I put a point about upgrading the whole chain on todays agenda.

korken89, 2019-03-12:
Perfect, I have been swamped with work and have not had time to do all the testing - I hope to get some time over soon to finalize this if no-one else has time to do it

mathk, 2019-03-14:
@rust-embedded/cortex-m Is this issue resolved now ?

korken89, 2019-03-14:
Yes, this is resolved

closed 2019-03-14 by korken89