rust-embedded/cortex-m #134: Waking up from wfe causes causes missed jump and hard fault

TheZoq2, 2019-02-17:
This is a very strange issue which I *think* I have tracked down to this crate. In my project, I have the following [code](https://github.com/TheZoq2/weather/blob/hardfault/stmhardware/src/main.rs#L397):

```rust
// Configure deep sleep
asm::wfe();
asm::nop(); // Added for debugging
```

This gets compiled to the following asm:

```asm
 8001d58:       5088            str     r0, [r1, r2]
 8001d5a:       f006 fb3a       bl      80083d2 <__wfe>
            __nop()
 8001d5e:       f006 fb34       bl      80083ca <__nop>
 8001d62:       6828            ldr     r0, [r5, #0]
```
```asm
080083ca <__nop>:
 80083ca:       4770            bx      lr

080083cc <__primask>:
 80083cc:       f3ef 8010       mrs     r0, PRIMASK
 80083d0:       4770            bx      lr

080083d2 <__wfe>:
 80083d2:       bf20            wfe
 80083d4:       4770            bx      lr

080083d6 <memcmp>:
 80083d6:       b5b0            push    {r4, r5, r7, lr}
 80083d8:       b382            cbz     r2, 800843c <memcmp+0x66>
 80083da:       f04f 0c03       mov.w   ip, #3
 80083de:       eb01 040c       add.w   r4, r1, ip
 80083e2:       eb00 050c       add.w   r5, r0, ip
 80083e6:       f814 ec03       ldrb.w  lr, [r4, #-3]
 80083ea:       f815 3c03       ldrb.w  r3, [r5, #-3]
 80083ee:       4573            cmp     r3, lr
```

- If the code runs normally, i.e. without any breakpoints, it hard faults at `80083ee`.
- Any breakpoints placed either after `1d5a` but before `1d62` or in __nop or
  `__wfe` makes the program run fine after continuing
- Stepping through one instruction or line at a time works and does not hard fault
- `84ce` and onwards are executed and have lr set to `1d67` which where the jump comes from.

To me, it looks like the branch at `83d4` is skipped but I'm not sure how to verify it.
I inspected the memory using gdb and instructions look correct

### Hardware and software

- This code is running on an stm32f103c8 blue pill. I have tested it on two different devices to ensure that it's not a hardware issue.
- It is 100% reproducible
- It happens both on rustc rustc `1.33.0-nightly (790f4c566 2018-12-19)` and `rustc 1.33.0-beta.6 (b203178b6 2019-02-05)`. I have not tested any other versions
- I'm using cortex-m version 0.5.8 from crates.io
- This does not seem to happen with `features = ["inline-asm"]`
- Lowering the `sysclk` seems to make it not happen every loop

This is the project where it happens :https://github.com/TheZoq2/weather/tree/hardfault/stmhardware.

I can't make a minimal example because the problem only occurs if some parts of the code are present. In this case, removing this line https://github.com/TheZoq2/weather/blob/hardfault/stmhardware/src/main.rs#L225 along with line 209 prevents it from happening.

Any ideas what could be going wrong here? I don't think I can do more with my limited knowlege of arm ASM

adamgreig, 2019-02-17:
Does the same problem happen if you use WFI instead of WFE? Are you specifically using events (i.e. calling SEV or waiting for exceptions)? Do you know what event/interrupt is causing the wakeup?

TheZoq2, 2019-02-18:
> Does the same problem happen if you use WFI instead of WFE?

Good question, I'm not sure. I'll test it later tody

> Are you specifically using events (i.e. calling SEV or waiting for exceptions)? Do you know what event/interrupt is causing the wakeup?

Yes, I'm using events to wake up from deep sleep as a way to conserve power. I'm using the RTC for triggerkng the wake up.

korken89, 2019-05-01:
Any more info on this? @TheZoq2 
I have now evaluated the same on an Nordic nRF52832 and STM32F411 and do not see this happening.

TheZoq2, 2019-05-05:
My fix for it was to use the inline-asm feature of this crate instead of the default features, but I'm not sure what caused the issue or why inline asm solves it.

korken89, 2019-05-06:
Unfortunate issue, I hope it was a one-timer but will keep this issue in mind for the future

closed 2019-08-13 by korken89

TheZoq2, 2019-08-31:
I think I may have found new information on this issue. I don't remember all the details of the issue, but there is a documented bug in the stm32f103 which sounds similar, see section 1.1.4 of https://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf