rust-embedded/cortex-m #137: Consider using Cargo's `links` feature to ensure use only once

jamesmunns, 2019-03-16:
Currently cortex-m uses an [unmangled variable](https://github.com/rust-embedded/cortex-m/blob/be6c39c725d2ae0cf50e36e07a27fd9a882cb71d/src/peripheral/mod.rs#L150-L153) to prevent multiple crates being linked in that redefine CORE_PERIPHERALS.

Cargo now has a feature called `links` that [addresses this](https://doc.rust-lang.org/cargo/reference/build-scripts.html#the-links-manifest-key).

little-arhat, 2019-04-23:
And how do I deal with this issue now? 

I've have dependency on `cortex-m`: `"=0.5.8"`, and dependency on `cortex-m-log`: `"0.4.0"` and everything worked fine for me, until suddenly some other crate brought `cortex-m 0.6.0` and linker started to complain: 

```
warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!

error: failed to load bc of "cortex_m.31bpwv5e-cgu.0":
```

looking at `Cargo.lock` I see that 0.6.0 is brough by `cortex-m-semihosting` (that `cortex-m-log` depends on):

```
[[package]]
name = "cortex-m-semihosting"
version = "0.3.3"
source = "registry+https://github.com/rust-lang/crates.io-index"
dependencies = [
 "cortex-m 0.6.0 (registry+https://github.com/rust-lang/crates.io-index)",
]

[[package]]
name = "cortex-m-log"
version = "0.4.0"
source = "registry+https://github.com/rust-lang/crates.io-index"
dependencies = [
 "cortex-m 0.5.8 (registry+https://github.com/rust-lang/crates.io-index)",
 "cortex-m-semihosting 0.3.3 (registry+https://github.com/rust-lang/crates.io-index)",
]
```

`cortex-m-log` defines `cortex-m` dependency as `cortex-m = ">= 0.5.8, < 0.7"`, and I have `cortex-m = {version = "=0.5.8", features = ["inline-asm", "const-fn"]}`, so I'm a bit surprised that `0.6.0` appeared in `Cargo.lock`.

jamesmunns, 2019-04-24:
@little-arhat this issue is a little different than what you are reporting, but I have run into this too.

I have had problems where despite the >=5.8,<0.7 stuff, cargo decided to pick cortex-m 0.5.8 AND 0.6. I manually edited the Cargo.lock and it was happy, but this doesn't seem optimal.

I even selected `cortex-m = "=0.5.8"` in my top level crate, but `cortex-m-semihosting` (a sub-sub-dep) still decided to pull in 0.6.

I "fixed" this locally by editing:

```
[[package]]
name = "cortex-m-semihosting"
version = "0.3.3"
source = "registry+https://github.com/rust-lang/crates.io-index"
dependencies = [
 "cortex-m 0.6.0 (registry+https://github.com/rust-lang/crates.io-index)",
]
```

to 

```
[[package]]
name = "cortex-m-semihosting"
version = "0.3.3"
source = "registry+https://github.com/rust-lang/crates.io-index"
dependencies = [
 "cortex-m 0.5.8 (registry+https://github.com/rust-lang/crates.io-index)",
]
```

Which seemed to work after a rebuild (and mentions of `cortex-m 0.6.0` were automatically removed by cargo), but this is sub-optimal behavior. CC @rust-embedded/cortex-m

little-arhat, 2019-04-24:
@jamesmunns thanks, that what I did as well, doesn't look pretty, indeed. 

I guess `links` feature will not help in this case, as it will _prevent_ linking, but will not alter solver behaviour (which is, I guess, responsible for the mess we're in %) ).

little-arhat, 2019-04-25:
@jamesmunns I've created cargo issue for this, btw: https://github.com/rust-lang/cargo/issues/6876

Eh2406, 2019-05-02:
BTW it links dose efact the resolver https://github.com/rust-lang/cargo/pull/4978

commit f1df6a8f (2019-04-24): Merge #140

140: Cargo.toml: Add `links = "cortex-m"` r=therealprof a=jonas-schievink

This prevents linking multiple versions of `cortex-m` together, which would be
unsound. Currently it uses a `#[no_mangle]` static for this, which isn't always
reliable.

Fixes #137
commit ddebec2b (2019-06-06): Merge #140

140: Cargo.toml: Add `links = "cortex-m"` r=therealprof a=jonas-schievink

This prevents linking multiple versions of `cortex-m` together, which would be
unsound. Currently it uses a `#[no_mangle]` static for this, which isn't always
reliable.

Fixes #137
PR #140 (2019-06-06): Cargo.toml: Add `links = "cortex-m"`
This prevents linking multiple versions of `cortex-m` together, which would be
unsound. Currently it uses a `#[no_mangle]` static for this, which isn't always
reliable.

Fixes #137
closed 2019-06-06 by bors[bot]