rust-embedded/cortex-m #146: 0.5.10 doesn't compile

thejpster, 2019-05-14:
If I try and compile 0.5.10 as part of nrf91-rs (0.1.0 is on crates.io) I get this error:

```
error: failed to run custom build command for `cortex-m v0.5.10`
process didn't exit successfully: `/home/jonathan/Documents/programming/testnrf/target/debug/build/cortex-m-91f41e6494ce18ce/build-script-build` (exit code: 101)
--- stderr
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: InvalidInput, error: StringError("the source path is not an existing regular file") }', src/libcore/result.rs:997:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
```
Is this a cortex-m fault or an something the nrf91-rs crate has done wrong?

little-arhat, 2019-05-14:
FWIF, I'm able to compile nrf91-rs with on both stable and nightly (2019-05-13) for my mac target and for `thumbv7em-none-eabihf`. Both `0.5.8` and `0.6.0` were compiled successfully during building of nrf crate.

jonas-schievink, 2020-01-12:
The error indicates a missing `bin/thumbv8-X.a`. We now do support ARMv8-M and those files exist, so closing.

closed 2020-01-12 by jonas-schievink