rust-embedded/cortex-m #148: SysTick exception not working for Blue Pill demo board (STM32f103C8)

Maldus512, 2019-06-16:
Hello everyone,
I was following the tutorials on the rust embedded book and got stuck with a problem on the exception example.
I'm using the exact code from the cortex-m-quickstart examples folder; It works on qemu (configured for the lm3s6965evb machine, but when I try running on real hardware the exception seems to cause some issues.
More specifically, I'm programming a Blue Pill demo board with a STM32f103C8 on it. To do so I changed the linker script (memory.x) to:

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* TODO Adjust these memory regions to match your device memory layout */
  /* These values correspond to the LM3S6965, one of the few devices QEMU can emulate */
  FLASH : ORIGIN = 0x08000000, LENGTH = 64K
  RAM : ORIGIN = 0x20000000, LENGTH = 20K
}
```
Plus openocd.cfg now uses stm32f1x.cfg.
Since the core is still a cortex-m3 I believe nothing else needs to be changed.
I run openocd, then arm-none-eabi-gdb; I can connect, load the program and enable semihosting just fine. The main reaches the endless loop but even with a breakpoint the exception seem to be never fired. Moreover, after the timer is reached even if I stop the execution the program counter reads an invalid address:
```
Program received signal SIGINT, Interrupt.
0x1ffff3b6 in ?? ()
```

This is the register dump:
```
r0             0x40010000          1073807360
r1             0xbfff              49151
r2             0x40003000          1073754112
r3             0xaaaa              43690
r4             0x0                 0
r5             0x799282            7967362
r6             0xe000e000          -536813568
r7             0x14                20
r8             0xffffffdc          -36
r9             0xf7ffffdf          -134217761
r10            0x431cf5a3          1125971363
r11            0xf6a31737          -157083849
r12            0x20000000          536870912
sp             0x200000c4          0x200000c4
lr             0x1ffff35f          536867679
pc             0x1ffff3b6          0x1ffff3b6
xPSR           0xa1000000          -1593835520   
msp            0x200000c4          0x200000c4
psp            0x2bac4eb0          0x2bac4eb0
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
```

What am I missing?

thejpster, 2019-06-16:
Could you try putting `cortex_m::asm::nop()` in the `loop { }` in `main`?
Are you compiling in Debug or Release? If you try the other, does that change anything?

Maldus512, 2019-06-17:
> Could you try putting cortex_m::asm::nop() in the loop { } in main?
> Are you compiling in Debug or Release? If you try the other, does that change anything?
I tried both (I was working in debug, tried compiling for release), but it yields the same results.

Maldus512, 2019-06-18:
I can confirm the exception example works on the nucleo-f030r8 (a cortex-m0), but still no luck on the blue pill demo board.

Disasm, 2019-06-18:
@Maldus512 Adding `unsafe { cortex_m::interrupt::enable() };` before the loop solved the problem for me.

TDHolmes, 2021-12-15:
@Maldus512 Did the above fix it for you? Seems like quite an old issue, perhaps it's time to close

closed 2021-12-17 by Maldus512