rust-embedded/cortex-m #149: [discussion] Peripheral singletons are not multi-core friendly

japaric, 2019-06-19:
## Background

### Private Peripheral Bus (PPB)

All the peripherals in the `cortex_m::Peripherals` struct sit on the Private
Peripheral Bus and have addresses of the form `0xE00x_xxxx`. Most of these
peripherals are interfaces to *internal* resources -- there's one instance of
those resources per core. Examples of internal PPB peripherals are the
`NVIC`, `MPU`, `SYST` (system timer), `ITM`, etc. The rest of peripherals are
interfaces to *external* resources meaning that all cores access the *same*
resources through these interfaces. An example of an external resource is the
`TPIU`.

The bottom line here is that interacting with a singleton like `CPUID` on one
core is different from interacting with it from another core even though they
are supposed to be the same singleton. For example, the `CPUID.base.read()`
operation can return different values depending on which core is executed.

### Multi-core API

In RTFM land we are exploring multi-core applications in two modes: homogeneous
mode and heterogeneous mode. Of course, it's very early days so we still don't
know what the ecosystem will adopt but these APIs let us show the problems with
the peripheral singletons.

#### Homogeneous

In this mode a binary crate is compiled using a single (compilation) target and
a single ELF image is produced. The image contains the entry points for all
cores and `static` variables have global visibility (visible to all cores) by
default.

``` rust
#![no_std]
#![no_main]

// visible to all cores
static X: AtomicBool = AtomicBool::new(false);

// core #0 user entry point
#[no_mangle]
unsafe extern "C" fn main_0() -> ! {
    // ..
}

// core #1 user entry point
#[no_mangle]
unsafe extern "C" fn main_1() -> ! {
    // ..
}
```

If this program is compiled for the `thumbv8m.main-none-eabi` then the resulting
image can be flashed, for example, on a 2x Cortex-M33 device.

#### Heterogeneous

In this mode a binary crate is compiled for *multiple* (compilation) targets so
multiple ELF images are produced. There's one image for each core and each image
contains the entry point and `static` variables for that core. There's a special
linker section named `.shared` used to *share* `static` variables between cores:
make them visible to all cores. One opts into this `.shared` section using the
`#[shared]` attribute; the default is that `static` variables are visible only
to the core where it was defined.

``` rust
#![no_std]
#![no_main]

// visible to all cores
[... 311 lines cut ...]

impl Timer {
    pub fn new() -> Self {
        // this operation effectively turns this type into an owned singleton
        SYST::seal();

        Self { _private: () }
    }

    pub fn set_timeout(&mut self, dur: Duration) {
         unsafe {
            SYST::borrow_unchecked(|syst| {
                // ..
            })
         }
    }
}
```

### Internal vs external resources

In the case of external resources like the `TPIU` I think we want to keep the
existing owned singleton / `take` API, with the semantics that only one core can
take these peripherals, because more than one core should not access these
resources in an unsynchronized fashion.

thejpster, 2019-06-22:
> AFAICT, there's no Cortex-M memory mapped register that returns a "core id"

Yeah, on the homogenous dual Cortex-M4 part of my Beagleboard X15 demo, I found the TI example using the Cortex-M4 Peripheral ID 0 register to determine which core was running. See https://github.com/cambridgeconsultants/rust-beagleboardx15-demo/blob/master/bare-metal/ipu-demo/src/main.rs#L435. Unfortunately it's vendor defined as to what they put in there :/ Perhaps it's something we can bring out in the chip crates, instead of being able to solve it for all Cortex-M cores.

The 'core local take' would work if the lock variable was in a core-specific address space, like the peripheral, right? So, like a thread-local variable.  I think the AM5728 had a block of RAM for each core, each mapped to the same address range, but I don't know if that's a universal thing or just an oddity of this implementation. You'd also need to ensure each core initialised its own memory correctly.

jonas-schievink, 2020-01-16:
I think the current implementations of `Send` are already correct. In standard Rust, `Send` models transferring data between *threads* within the same process (which have access to all of its resources), which is akin to moving between interrupt handlers running on the same core, or between cores that share a single set of resources (that is, all memory and peripherals).

Since most (all?) multi-core MCUs have core-local peripherals or memory, transferring objects between the cores is more similar to cross-*process* communication in standard Rust and should not be modeled with `Send`, but a custom trait.