rust-embedded/cortex-m #154: Need some help with interrupts

josephpenafiel, 2019-07-10:
Hello everyone, 

I'm trying to test some interrupts in a stm32f302 nucleo board, but I'm having some trouble with the Mutex. When I define one, I get this error: 
"calls in statics are limited to constant functions, tuple structs and tuple variants rustc (E0015)"

here's the code I'm trying to test: 

`#![no_main]
#![no_std]
#[allow(unused_extern_crates)]

extern crate panic_halt;
extern crate stm32f30x;

use core::cell::RefCell;
use cortex_m_rt::entry;
// use cortex_m::interrupt::{Mutex, self}; 
use stm32f30x::{interrupt,Interrupt};

    static MY_TIM2: cortex_m::interrupt::Mutex<RefCell<Option<stm32f30x::TIM2>>> = cortex_m::interrupt::Mutex::new(RefCell::new(None)); // <----error
    static mut TOOGLE: bool = false;

#[entry]
fn main() -> ! {
    let p = cortex_m::Peripherals::take().unwrap();
    let cp = stm32f30x::Peripherals::take().unwrap();
    let rcc = &cp.RCC;
    let gpiob = &cp.GPIOB;
    let tim2 = &cp.TIM2;

    rcc.apb1enr.write(|w| w.tim2en().set_bit()); // tim2 clock
    rcc.ahbenr.write(|w| w.iopben().set_bit()); // pb2 clock
    
    gpiob.moder.write(|w| w.moder13().output());
    gpiob.odr.write(|w| w.odr13().set_bit());

    tim2.psc.write(|w| unsafe {w.psc().bits(23_999)}); // 24k prescaler
    tim2.arr.write(|w| unsafe {w.arrl().bits(1000)}); // auto reload value
    tim2.dier.write(|w| w.uie().set_bit());
    tim2.cr1.write(|w| w.cen().set_bit());

    // interrupt::free(|cs| MY_GPIO.borrow(cs).replace(Some(cp.GPIOB)));
    cortex_m::interrupt::free(|cs| MY_TIM2.borrow(cs).replace(Some(cp.TIM2)));

    let mut nvic = p.NVIC;
    nvic.enable(Interrupt::TIM2);

    loop {
        
    }
}

#[interrupt]
fn TIM2() {
    cortex_m::interrupt::free(|cs| {
        let tim2 = MY_TIM2.borrow(cs).borrow();
        if tim2.as_ref().unwrap().sr.read().uif().bit_is_set() {
            unsafe { TOOGLE = !TOOGLE };
            tim2.as_ref().unwrap().sr.write(|w| w.uif().clear_bit());
        }
    });

}`

I followed the example from the Rust embedded book in the concurrency section. 

thanks in advance

geomatsi, 2019-07-10:
Hi @josephpenafiel 
In brief, it should be enough to enable `const-fn` feature in `cortex-m` crate:

```
[dependencies.cortex-m]
version="0.6.0"
features=["const-fn"]
```

This feature is going to be enabled by default some time in the future: see #153 .  As for the embedded-rust book, the appropriate update is on review at the moment: see [PR#196](https://github.com/rust-embedded/book/pull/196).

Regards,
Sergey

josephpenafiel, 2019-07-10:
thank you that worked, but now I have a new problem. I want to do something simple. Toggle a led everytime a timer interrupt occurs. 
I made a Mutex for the timer, since I need to access it in the interrupt handler to clear the interrupt flag. But since all the peripherals moved into the Mutex, I can't access the GPIO anymore.

The compiler tells me this:

     let gpiob = &cp.GPIOB;
   |                 --------- borrow of `cp.GPIOB` occurs here
...
44 |     cortex_m::interrupt::free(|cs| MY_TIM2.borrow(cs).replace(Some(cp.TIM2)));
   |                               ^^^^ move out of `cp` occurs here    -- move occurs due to use in closure
...
57 |                 gpiob.odr.write(|w| w.odr13().clear_bit());
   |                 ----- borrow later used here

what's the correct way of doing what I want?

geomatsi, 2019-07-10:
I would suggest to try the following approach: make both TIM2 and GPIO (or LED pin) global shared resources. Then use them in TIM2 handler under interrupt::free mutex.

josephpenafiel, 2019-07-10:
when i do that I get a similar error
`
 --> src/main.rs:47:31
   |
27 |     let cp = stm32f30x::Peripherals::take().unwrap();
   |         -- move occurs because `cp` has type `stm32f30x::Peripherals`, which does not implement the `Copy` trait
...
46 |     cortex_m::interrupt::free(|cs| MY_GPIOB.borrow(cs).replace(Some(cp.GPIOB)));
   |                               ---- value moved into closure here    -- variable moved due to use in closure
47 |     cortex_m::interrupt::free(|cs| MY_TIM2.borrow(cs).replace(Some(cp.TIM2)));
   |                               ^^^^ value used here after move      -- use occurs due to use in closure
`

geomatsi, 2019-07-10:
Hi @josephpenafiel 

Here is an example for BluePill that converts both TIM2 and PC13 into shared resources and uses them in a safe way: [safe LED blink from TIM2 irq](https://github.com/geomatsi/rust-blue-pill-tests/blob/master/src/bin/blink-timer-irq-safe.rs).

Regards,
Sergey

josephpenafiel, 2019-07-13:
not helpful

closed 2019-07-13 by josephpenafiel