rust-embedded/cortex-m #155: thumbv8m.base +inline-asm  = "error: invalid operand for instruction"

japaric, 2019-07-11:
### STR

Grab the cortex-m-quickstart template (a version that uses cortex-m v0.6.0) and compile this program:

``` rust
#[entry]
fn main() -> ! {
    let x = basepri::read();

    loop {}
}
```

for the `thumbv8m.base-none-eabi` target but make sure that the "inline-asm" feature of cortex-m has been enabled; you get:

``` console
error: <inline asm>:1:10: error: invalid operand for instruction
        mrs r0, BASEPRI
                ^
```

The BASEPRI register doesn't exist on ARMv8-M devices that don't have the main extension. The `register::{basepri,basepri_max,faultmask}` APIs should not be exposed for the `thumbv8m.base-none-eabi` target.

As removing an API is a breaking-change (it breaks libraries; the above error only occurs in applications) I would suggest marking the API as deprecated on the `thumbv8m.base-none-eabi` target and removing it in the next minor version.

This API should *not* be removed from the `thumbv8m.main-none-eabi*` targets (note: *main*, not *base*) as it is supported on those devices.

### Metadata

``` console
$ rustc -V
rustc 1.38.0-nightly (cd2cd4c96 2019-07-10)
```

---

I also noted that the `thumbv8m.base-none-eabi` target is not being tested on Travis and that it doesn't have special handing (`cargo:rustc-cfg`) in the `build.rs` so there are may be more errors similar to this one in the crate.

JJJollyjim, 2019-09-20:
> so there are may be more errors similar to this one in the crate

The crate certainly compiles once I configure out the basepri stuff.

> As removing an API is a breaking-change (it breaks libraries...

I can certainly vouch that it breaks rtfm :)

korken89, 2019-10-29:
I think this can be closed now, reopen if the issue persists

closed 2019-10-29 by korken89