rust-embedded/cortex-m #156: thumbv8.main -inline-asm = "undefined symbol: __basepri_r"

japaric, 2019-07-11:
### STR

Start from the cortex-m-quickstart template and write:

``` rust
#[entry]
fn main() -> ! {
    let x = basepri::read();

    loop {}
}
```

compile for the `thumbv8m.main-none-eabi` target; you get:

``` console
error: linking with `rust-lld` failed: exit code: 1
  = note: rust-lld: error: undefined symbol: __basepri_r
```

The `thumbv8m.main-none-eabi.a` archive / binary blob is missing `__basepri_r` and other symbols. `asm-v7.s` should be compiled for this target and included in the archive.

### Metadata

``` console
$ rustc -V
rustc 1.36.0 (a53f9df32 2019-07-03)
```

JJJollyjim, 2019-09-20:
I just experienced this after updating to the latest master `cortex-m-rtfm`, on `thumbv8m.base-none-eabi`

JJJollyjim, 2019-09-20:
Aah, apparently that's because "In a PE with the Main Extension, software can use the PRIMASK, FAULTMASK, and BASEPRI registers as follows. A PE without the Main Extension implements PRIMASK, but does not implement FAULTMASK and BASEPRI"... ignore what I said

JJJollyjim, 2019-09-20:
I believe this was resolved as part of #161

korken89, 2019-10-29:
This should be solved now

closed 2019-10-29 by korken89