rust-embedded/cortex-m #157: thumbv8m.main-none-eabihf = "No such file or directory"

japaric, 2019-07-11:
### STR

Grab the `cortex-m-quickstart` template and compile it for `thumbv8m.main-none-eabihf`; you get:

``` console
error: failed to run custom build command for `cortex-m-rt v0.6.9`
--- stderr
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/libcore/result.rs:999:5
```

repeated three times. There's no `thumbv8m.main-none-eabihf.a` archive in this crate (or in cortex-m{,-rt,-semihosting}) yet the build script tries to copy the archive.

### Metadata

``` console
$ rustc -V
rustc 1.36.0 (a53f9df32 2019-07-03)
```

thejpster, 2019-07-14:
New targets will continue to appear, so maybe we should fix the build.rs to be more prescriptive about which targets are supported, and give a better error of you use one we don't haven't prebuilt objects for?

aurabindo, 2019-08-10:
The non existing archive is not generated in assemble.sh for `thumbv8m.main-none-eabihf`. Is there any particular reason ? I'm wondering why archive for hardfloat wasnt added in ac63dee16cb106228e15f9d9c710bf91b832a4f8

aurabindo, 2019-08-13:
@thejpster  I believe this issue can be closed now, since 57d44f6648bba3eb2746e64fbf28cbac4012a19c takes care of this error.

nickray, 2019-08-21:
I think some releases are in order before this can be closed.

jonas-schievink, 2020-01-14:
We've recently released cortex-m 0.6.2, so closing this.

closed 2020-01-14 by jonas-schievink