rust-embedded/cortex-m #166: Read Access to control regiater

kanishkarj, 2019-09-15:
Hi ! I need access to read from control register. Basically i want to check if the code is currently running in privileged or unprivileged mode. But the thing being, how to initialize an instance of control register? (Cotrex_m:: registers::control::Control)

jonas-schievink, 2019-09-27:
`cortex_m::register::control::read()`

kanishkarj, 2019-11-02:
I used that and it was working fine for a while. Then suddenly it started behaving weirdly, despite switching to user mode and to PSP (in an RTOS) the control register showed that the program is currently running in privileged mode and MSP. After debugging for a long time, I tried reading values from the control register directly via inline-asm. Any reason why it happened so? (currently I am using the inline-asm function instead of the one provided by cortex-m crate)

jonas-schievink, 2020-01-14:
I don't think there's much we can do here, especially without source code. Reading `CONTROL` works fine for me.

closed 2020-01-14 by jonas-schievink