rust-embedded/cortex-m #170: Controller Area Network (CAN bus)

reneherrero, 2019-10-05:
Is CAN bus communications something that is supported in rust on Cortex M?  Any getting started tips/links?

All I'm finding in Google is based on Linux/SocketCAN.  I can't imagine I'm the first to have this need?

Apologies if this is not the appropriate channel, I've recently fallen in love with Rust and am venturing into new territories...  C is a dinosaur that needs to be turned into oil  :)

hannobraun, 2019-10-07:
The `cortex-m` crate only covers the Cortex-M core peripherals, while CAN peripherals would be vendor-specific. So you're asking in the wrong project :-)

[`embedded-hal`](https://crates.io/crates/embedded-hal) provides an abstraction layer for vendor-specific peripherals. It doesn't support CAN as of yet, but there's been discussion and some stalled work: https://github.com/rust-embedded/embedded-hal/issues/21, https://github.com/rust-embedded/embedded-hal/pull/53, https://github.com/rust-embedded/embedded-hal/pull/77

In addition, there might already be HAL APIs that support CAN for a specific chip. I can't think of any off the top of my head though, but maybe you can find something in the [stm32-rs organization](https://github.com/stm32-rs).

reneherrero, 2019-10-07:
Thanks for the info @hannobraun, and apologies for the noise.  Closing the issue.

closed 2019-10-07 by reneherrero