rust-embedded/cortex-m #176: [Question] NVIC_SetPriorityGrouping?

Geobert, 2019-10-31:
Bear with me, my knowledge of embedded world it still WIP.

Working on a Silab chip, I tend to read the C code of examples and try to do the equivalent the Rust way.

I encounter a function called `NVIC_SetPriorityGrouping` doing some stuff in `SCB->AIRCR`. It's in a file called `cortex_m4.h".

Is there the equivalent in this crate? I can't find anything in the code nor the doc.

thejpster, 2019-11-02:
C docs for that function: https://www.keil.com/pack/doc/cmsis/Core/html/group__NVIC__gr.html#gad78f447e891789b4d8f2e5b21eeda354

Possible C implementation for that function: https://github.com/hrshygoodness/EFM32-Library/blob/a98a6bd681454da0ad86a048c0f28389c5c798e7/v2/CMSIS/Include/core_cm3.h#L1291

Register definitions are on page 4-16 in this ARM Cortex-M PDF: http://infocenter.arm.com/help/topic/com.arm.doc.dui0553b/DUI0553.pdf

Unfortunately we don't have a nice wrapper like CMSIS does at this time. I think our equivalent would be manually setting a value in scb.aircr (https://docs.rs/cortex-m/0.6.1/cortex_m/peripheral/scb/struct.RegisterBlock.html#structfield.aircr)

Geobert, 2019-11-02:
Thank you for your answer, it was what I thought :)

closed 2019-11-02 by Geobert