rust-embedded/cortex-m #177: [Question] nvic::set_priority against NVIC_SetPriority

Geobert, 2019-11-08:
Hi,

It's me again :)

I'm looking at the code of `ncvic::set_priority` and comparing it to the C impl that I got from my supplier and they are quite different:
```C
__STATIC_INLINE void __NVIC_SetPriority(IRQn_Type IRQn, uint32_t priority)
{
  if ((int32_t)(IRQn) >= 0)
  {
    NVIC->IP[((uint32_t)IRQn)] = (uint8_t)((priority << (8U - __NVIC_PRIO_BITS)) & (uint32_t)0xFFUL);
  }
  else
  {
    SCB->SHP[(((uint32_t)IRQn) & 0xFUL)-4UL] = (uint8_t)((priority << (8U - __NVIC_PRIO_BITS)) & (uint32_t)0xFFUL);
  }
}
```

Writing even in different peripherical. Why are they so different?

korken89, 2019-11-12:
They do the same, only that the Rust implementation had split SCB and NVIC properties.

See here: https://docs.rs/cortex-m/0.6.1/cortex_m/?search=set_prio

Geobert, 2019-11-13:
I see, so if priority is < 0, I call SCB::set_priority. Thank you !

closed 2019-11-13 by Geobert

korken89, 2019-11-13:
If you have the cortex core interrupts use SCB, device interrupts use NVIC :) priority works the same for both

Geobert, 2019-11-13:
As you can see, the SDK I'm using encode the core one in negative `int32_t` and to some C trick to convert them into proper index. I need to convert these into `SystemHandler` but I got the idea, thank you very much!

Geobert, 2019-11-13:
Last question on this matter: I can see that `NonMaskableInt` and `HardFault` are commented out because priority is fixed.

What happen in C code then if somebody tries to set their priority anyway?
Asking because they are declared in the enum of the SDK:
```C
NonMaskableInt_IRQn   = -14,              /*!< 2  Cortex-M4 Non Maskable Interrupt      */
  HardFault_IRQn        = -13,              /*!< 3  Cortex-M4 Hard Fault Interrupt        */
```

korken89, 2019-11-13:
There is a good chance it would hardfault or just cause UB, not sure though.