rust-embedded/cortex-m #183: Ship a 1.0 release

thejpster, 2020-01-07:
Part of https://github.com/rust-embedded/wg/issues/383 

Might also involve https://github.com/rust-lang/rust/issues/27731 so we can get rid of (some) of the inline asm / pre-compiled objects.

jonas-schievink, 2020-01-07:
> Might also involve [rust-lang/rust#27731](https://github.com/rust-lang/rust/issues/27731) so we can get rid of (some) of the inline asm / pre-compiled objects.

This is not needed for 1.0 since it doesn't change the public API

jonas-schievink, 2020-01-09:
Easy (breaking) improvements to do before 1.0:
* Remove the `const-fn` feature
* Get https://github.com/rust-embedded/cortex-m/pull/181 landed and audit the crate for other things that should be `#[cfg]`d out depending on the target

Other 1.0-quality things that I'd like to see:
* Improve overall documentation
* Document that the `inline-asm` feature is unstable (semver-exempt) and nighly-only

This is also blocked on sound MMIO. Currently we use `volatile-register` here. As #184 mentions, there's also a public dependency on `aligned` that we should get rid of (or also bring to 1.0).

adamgreig, 2020-01-09:
I'd really want to see cortex-m move to using svd2rust-generated API for accessing the Cortex-M peripherals, i.e. a cortex-m-pac crate that cortex-m depends on and adds the HAL API. I've made some progress towards generating a suitable set of SVD files already and plan to finish it fairly soon.

thejpster, 2020-01-09:
> 
> 
> I'd really want to see cortex-m move to using svd2rust-generated API for accessing the Cortex-M peripherals, i.e. a cortex-m-pac crate that cortex-m depends on and adds the HAL API. I've made some progress towards generating a suitable set of SVD files already and plan to finish it fairly soon.

Yeah, this would likely fundamentally change our public API so if we are ever going to do it, pre 1.0 is best.

jonas-schievink, 2020-01-10:
Also:
* Remove currently `#[deprecated]` unsound APIs (I expect we'll publish 0.7 before 1.0, so it would be good to do all breaking changes there)

adamgreig, 2020-01-28:
I'd add:

* Check all peripheral methods that could be static are, i.e. don't unnecessarily require a reference to the peripheral for safe/atomic operations. We've had a number of breaking changes moving things from methods to associated functions and the latter are easier to use, so we should proactively see if we can do any more.

Disasm, 2020-02-08:
With the [new CriticalSection type](https://github.com/rust-embedded/bare-metal/pull/20) `interrupt::free()` should pass `CriticalSection` by value instead of by reference.

tmplt, 2022-01-04:
@adamgreig any updates on the svd2rust generation? Is there a tracking issue for it?

Gryhyphen, 2024-11-03:
So this is embeded-hal 1.0 incompatibility? Was working on updating my dependencies and found that cortext-m was relying on `embedded-hal v0.2.7`, so having `embdedded-hal v1.0.0` is not possible if we want to use cortext-m atm? If I'm understanding correctly?

adamgreig, 2024-11-03:
cortex-m currently uses embedded-hal 0.2, but it's not a problem to have both embedded-hal 0.2 and 1.0 in a project at the same time. That way you can use embedded-hal 1.0 and all the drivers and HALs that use it at the same time as using cortex-m.

adamgreig, 2025-08-06:
Hijacking this issue to track things we planned to make as breaking changes in 0.8 but were reverted by #605 

- Remove prelude #425 
- Remove deprecated `ptr()` #385 
- VECTACTIVE should be 9 bits not 8 #373 
- don't require `&mut self` for nvic request #472