rust-embedded/cortex-m #184: Consider getting rid of aligned dependency subtree

nickray, 2020-01-08:
This eventually pulls in two versions of generic-array which is a bit weird for just one method. Maybe take over aligned or as-slice and add some feature flags there? Or make `itm::write_aligned` itself an optional feature?

re. https://github.com/rust-embedded/cortex-m/issues/183

jonas-schievink, 2020-01-14:
Oh *this* is where all the `generic-array` dependencies come from? I was wondering about that.

Pulling in 4 dependencies for a single function seems very excessive, so I'm on board with removing them. We can add our own alignment-enforcing struct to the `itm` module instead.

nickray, 2020-01-14:
Yeah for some (compatibility?) reason as-slice pulls in both.

therealprof, 2020-01-14:
Yeah, otherwise you run into weird compatibility problems.

nickray, 2020-01-14:
None of my dependencies that depend on generic-array use 0.12, so I' be curious what compatibility issues there might be?

closed 2020-01-15 by bors[bot]