rust-embedded/cortex-m #188: "Invalidate Cache" operation is unsafe

jonas-schievink, 2020-01-13:
The optional Cortex-M cache exposes "clean" and "invalidate" operations. As pointed out by @adamgreig in https://github.com/rust-embedded/cortex-m/issues/47#issuecomment-573482398, performing only the "invalidate" operation without a preceding "clean" is unsound.

* "Clean" flushes cache lines to main memory and makes operations visible to other peripherals and cores
* "Invalidate" drops cache lines so that they are fetched from main memory on subsequent accesses, making modifications from other bus masters visible to the core doing the invalidation

This operation is currently exposed via safe interfaces:
* As methods on [`CBP`](https://docs.rs/cortex-m/0.6.2/cortex_m/peripheral/struct.CBP.html)
* As methods on [`SCB`](https://docs.rs/cortex-m/0.6.2/cortex_m/peripheral/struct.SCB.html)

We should either deprecate the methods and add unsafe replacements, or make the methods `unsafe` in a breaking change without deprecation period (this avoids having to find new names).

adamgreig, 2020-01-13:
> performing only the "invalidate" operation without a preceding "clean" is unsound

It might also be that a clean is unsound too, I want to think about this a little more.

> make the methods unsafe in a breaking change without deprecation period

Given the planned upcoming breaking changes and I think the relatively uncommon use of the cache operations this is my preferred option.

> In my opinion there is little to no value in exposing only "invalidate" since it hard to use correctly and has little to no benefits over "clean + invalidate", so removing it is also an option.

These operations are peripheral registers in Cortex-M, with clean+invalidate being distinct from invalidate - sometimes you need one and sometimes you need the other. The methods in SCB mirror those provided in CMSIS. You very often need to invalidate without cleaning if you know another observer has written to the main memory and your cache is outdated, since a clean+invalidate would overwrite whatever the observer had written. Clean+invalidate is pretty much just used for disabling the cache, when you want to ensure all updates are written out and then the item is removed from the cache. I definitely think we should keep all the methods.

jonas-schievink, 2020-02-05:
Fixed by https://github.com/rust-embedded/cortex-m/pull/192, closing

closed 2020-02-05 by jonas-schievink