rust-embedded/cortex-m #195: UB in inline-asm delay function

ButtNaked, 2020-02-11:
UB in higher opt levels with **inline-asm feature**, caused by forgotten asm clobber in asm::delay function.

Example fix:
```            
asm!("1:
      nop
      subs $0, $$1
      bne.n 1b"
      : "+r"(_n / 4 + 1)
      :
      :
      : "cpsr" // <= Added here
      : "volatile");
```

jonas-schievink, 2020-02-11:
Oh, you need to clobber the status register explicitly? That seems very easy to forget, we should audit all asm for this.

adamgreig, 2020-02-18:
I've done an audit of all occurrences of `asm!` I could find in rust-embedded and this is the only instance I found which updates the condition flags. Should the fix be `cpsr` or `cc` clobber?

therealprof, 2020-02-18:
`cspr` should be the right one since it's the name of the register containing the condition flags on ARM. `cc` is some GCC architecture independent magic name to indicate the register holding the condition flags, but I'm not even sure it's universally applicable to non-GCC inline assembly...

commit e106d4a8 (2020-03-15): Add CPSR to asm::delay's clobber flags. Fixes #195
commit 70be119f (2020-03-15): Merge #200

200: Add CPSR to asm::delay's clobber flags. Fixes #195 r=therealprof a=adamgreig
commit 8c5030eb (2020-03-15): Merge #200

200: Add CPSR to asm::delay's clobber flags. Fixes #195 r=therealprof a=adamgreig
commit 1830823a (2020-07-19): Add CPSR to asm::delay's clobber flags. Fixes #195
closed 2020-03-15 by bors[bot]