rust-embedded/cortex-m #217: v0.6.3 release?

japaric, 2020-05-19:
Did anything breaking land in master since the v0.6.2 release?

I would like to see a v0.6.x release that includes binary-level fixes like #212 and #216

Disasm, 2020-05-19:
There was a breaking change since v0.6.2: https://github.com/rust-embedded/cortex-m/pull/190

little-arhat, 2020-06-07:
0.6.2 no longer works in nightly (`rustc 1.46.0-nightly (118b50524 2020-06-06)`) due to `asm!` deprecation (was fixed in https://github.com/rust-embedded/cortex-m/issues/204).

jonas-schievink, 2020-06-07:
It should still work without the unstable `inline-asm` feature

little-arhat, 2020-06-07:
@jonas-schievink 

```
When this feature is enabled the implementation of all the functions inside the asm and register modules use inline assembly (asm!) instead of external assembly (FFI into separate assembly files pre-compiled using arm-none-eabi-gcc). The advantages of enabling inline-asm are:

    Reduced overhead. FFI eliminates the possibility of inlining so all operations include a function call overhead when inline-asm is not enabled.

    Some of the register API only becomes available only when inline-asm is enabled. Check the API docs for details.

The disadvantage is that inline-asm requires a nightly toolchain.
```

Documentation doesn't say this feature is unstable.

adamgreig, 2020-07-05:
I think we should start looking in to a 0.6.3 release with only the non-breaking changes and probably a 0.7.0 series with the breaking changes; as far as I'm aware the breaking PRs since 0.6.2 are #180 #181 #190 #191 #192 #193.

kanishkarj, 2020-07-19:
It would be really great if `0.6.3` comes out soon. I want to actually publish a crate but the docs don't build because it uses the latest nightly. And it is not possible on the latest nightly like @little-arhat pointed out.

jonas-schievink, 2020-07-19:
I'll take this on

adamgreig, 2020-07-20:
0.6.3 is now published :tada:

closed 2020-07-20 by adamgreig

kanishkarj, 2020-07-20:
Thanks a lot!!

hug-dev, 2020-07-20:
Thanks for the release!

A small question: the `cmse` module does not seem to appear in the [docs](https://docs.rs/cortex-m/0.6.3/cortex_m/). Is that because it is behind the `armv8m` feature?

adamgreig, 2020-07-20:
Ah, that's annoying, yes that will be why. We should add a docs.rs metadata section to our `Cargo.toml` and perhaps a new `doc` feature to enable all modules when building for docs.rs. Though in some cases we include conflicting structs depending on architecture, so only one of those can be shown to docs.rs, which is kind of a pain...

hug-dev, 2020-07-20:
Maybe showing the latest and greatest architecture is a good idea as it will most probably include all previous architecture features? For example that would be `armv8m_main` and `has_fpu` today?
The bad thing is that it will also show code that is not compiled for some architectures but imo that is better than not showing the code compiled in some others 😃.