rust-embedded/cortex-m #223: Unable to create a MonoTimer

arturjpv, 2020-06-07:
# Error Description

I'm trying to create a MonoTimer and the code is not compiling.

I'm receiving a message that could point to mixed crates or incompatibility in the dependencies.
Perhaps `cortex-m v0.6.2` and `cortex-m v0.5.10`? I don't understand how is that `cargo tree` is listing `cortex-m v0.6.2` as a dependency of `cortex-m v0.5.10`

```
error[E0308]: mismatched types
  --> src/core/mod.rs:27:37
   |
27 |         let chrono = MonoTimer::new(cp.DWT, clocks);
   |                                     ^^^^^^ expected struct `cortex_m::peripheral::DWT`, found a different struct `cortex_m::peripheral::DWT`
   |
   = note: perhaps two different versions of crate `cortex_m` are being used?
```

# Dependencies
The important parts of my Cargo.toml look like this:
```
[dependencies]
cortex-m = "0.6.2"
cortex-m-rt = "0.6.12"
panic-halt = "0.2.0"

[dependencies.f3]
features = ["rt"]
version = "0.6.1"
```

Output of `cargo tree`:
```
chip8stm v0.1.0 (/home/artur/Documents/Code/chip8/chip8stm)
├── chip8vm v0.8.2 (/home/artur/Documents/Code/chip8/chip8vm)
├── cortex-m v0.6.2
│   ├── aligned v0.3.2
│   │   └── as-slice v0.1.3
│   │       ├── generic-array v0.12.3
│   │       │   └── typenum v1.12.0
│   │       ├── generic-array v0.13.2
│   │       │   └── typenum v1.12.0
│   │       └── stable_deref_trait v1.1.1
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   └── volatile-register v0.2.0
│       └── vcell v0.1.2
├── cortex-m-rt v0.6.12
│   ├── cortex-m-rt-macros v0.1.8
│   │   ├── proc-macro2 v1.0.18
│   │   │   └── unicode-xid v0.2.0
│   │   ├── quote v1.0.6
│   │   │   └── proc-macro2 v1.0.18 (*)
│   │   └── syn v1.0.30
│   │       ├── proc-macro2 v1.0.18 (*)
│   │       ├── quote v1.0.6 (*)
│   │       └── unicode-xid v0.2.0
│   └── r0 v0.2.2
├── f3 v0.6.1
│   ├── l3gd20 v0.2.0
│   │   ├── embedded-hal v0.2.3
│   │   │   ├── nb v0.1.2
│   │   │   └── void v1.0.2
│   │   └── generic-array v0.11.1
│   │       └── typenum v1.12.0
│   ├── lsm303dlhc v0.2.0
│   │   ├── cast v0.2.3
│   │   │   [build-dependencies]
│   │   │   └── rustc_version v0.2.3 (*)
│   │   ├── embedded-hal v0.2.3 (*)
│   │   └── generic-array v0.11.1 (*)
│   └── stm32f30x-hal v0.2.0
│       ├── cast v0.2.3 (*)
│       ├── cortex-m v0.5.10
│       │   ├── aligned v0.2.0
│       │   ├── bare-metal v0.2.5 (*)
│       │   ├── cortex-m v0.6.2 (*)
│       │   └── volatile-register v0.2.0 (*)
│       ├── embedded-hal v0.2.3 (*)
│       ├── nb v0.1.2
│       ├── stm32f30x v0.7.1
│       │   ├── bare-metal v0.2.5 (*)
│       │   ├── cortex-m v0.5.10 (*)
│       │   ├── cortex-m-rt v0.6.12 (*)
│       │   └── vcell v0.1.2
│       └── void v1.0.2
└── panic-halt v0.2.0
```

# Additional information
- Compiling on stable channel of rust 1.44.0.
- I'm developing for the stm32f3Discovery board (stm32f303).
- Target system is thumbv7em-none-eabihf.

arturjpv, 2020-06-08:
I finally could create de MonoTimer using:
```rust
f3::hal::stm32f30x::CorePeripherals
```
instead of
```rust
cortex_m::Peripherals
```

But now I don't understand the difference between 'CorePeripherals' and 'CortexPeripherals'.

jonas-schievink, 2021-01-08:
`stm32f30x-hal` pulls in cortex-m version 0.5.x, while you were trying to use 0.6.x, as suggested by the error message

closed 2021-01-08 by jonas-schievink