rust-embedded/cortex-m #224: `Mutex::borrow` only gives an immutable reference?

jamadazi, 2020-06-08:
It seems like the `cortex_m::interrupt::Mutex` type only gives an immutable reference? What's the point of the mutex then? How can I get a mutable reference to the value?

jonas-schievink, 2020-06-08:
`Mutex` makes `Send` data `Sync`. You can use cell-based interior mutability inside it.

The reason it is done this way is because the user knows best which cell type to use (eg. `RefCell` is more costly than `Cell` but allows obtaining references to the inner value).

Wassasin, 2020-06-09:
You can nest two `CriticalSections`, possibly yielding multiple instances of `&mut T` if `borrow_mut` were possible. You can use `Mutex<UnsafeCell<T>>` and get a mutable reference that way if you take care to never nest the `CriticalSections`.

TDHolmes, 2021-12-15:
Related: #204