rust-embedded/cortex-m #229: iprintln!() halts printing after printing certain amount of data

JOE1994, 2020-06-12:
Hello :crab:, when running the program below that repeatedly calls `iprintln!()`, it always stops printing at the same point in code, not finishing the printing tasks. It should print "Hello World ~" 77 times in a loop, but the program always stops printing in the middle of 26th invocation of `iprintln!()`.
 
```rust
#[entry]
fn main() -> ! {
    if let (Some(dp), Some(mut cp)) = (
        stm32::Peripherals::take(),
        cortex_m::peripheral::Peripherals::take(),
    ) {
        // Set up the system clock. We want to run at 168MHz for this one.
        let rcc = dp.RCC.constrain();
        let clocks = rcc.cfgr.sysclk(168.mhz()).freeze();
        let mut itm = cp.ITM;
        for i in 0..77 {
            iprintln!(&mut itm.stim[0], "Hello World for the {}-th time", i);            
        }

        loop {
            continue;
        }
    }

    // Failed to access peripherals..
    panic!();
}
```
logs from ITM are stored locally in **itm.txt**, and the log is always the same as below
(no matter how many times I try running).
```txt
Hello World for the 0-th time
Hello World for the 1-th time
Hello World for the 2-th time
Hello World for the 3-th time
Hello World for the 4-th time
Hello World for the 5-th time
Hello World for the 6-th time
Hello World for the 7-th time
Hello World for the 8-th time
Hello World for the 9-th time
Hello World for the 10-th time
Hello World for the 11-th time
Hello World for the 12-th time
Hello World for the 13-th time
Hello World for the 14-th time
Hello World for the 15-th time
Hello World for the 16-th time
Hello World for the 17-th time
Hello World for the 18-th time
Hello World for the 19-th time
Hello World for the 20-th time
Hello World for the 21-th time
Hello World for the 22-th time
Hello World for the 23-th time
Hello World for the 24-th time
Hello World
```
Is there a maximum limit of data that `iprintln!` can send in total?
If not, I guess this should be considered a bug..

## Metadata
* I used `stm32f407g-disc1` board to run this program.
* `rustc --version --verbose`
```
rustc 1.46.0-nightly (feb3536eb 2020-06-09)
binary: rustc
commit-hash: feb3536eba10c2e4585d066629598f03d5ddc7c6
commit-date: 2020-06-09
host: x86_64-unknown-linux-gnu
release: 1.46.0-nightly
LLVM version: 10.0
```

bcantrill, 2020-06-12:
How are you consuming ITM? For whatever it's worth, your example works for me with an ITM consumer that we have purpose built that consumes ITM via the TPIU; are you using the STLink/V2 on the Discovery and then OpenOCD? And is your program stopping, or is the ITM data being dropped?

JOE1994, 2020-06-12:
> How are you consuming ITM? For whatever it's worth, your example works for me with an ITM consumer that we have purpose built that consumes ITM via the TPIU; are you using the STLink/V2 on the Discovery and then OpenOCD? And is your program stopping, or is the ITM data being dropped?

Thank you so much for testing the example on your side! :smile_cat: 
I am using STLink/V2-1 on the Discovery and with OpenOCD.
The program runs to completion and reaches the infinite loop at the end,
and only the ITM data is dropped :crying_cat_face: 

(Using `gdb-multiarch`),
I tried executing the code instruction-by-instruction using the `n` command
to see if doing that could make the program successfully run all invocations of `iprintln!()`..
But pressing `n` just once simply executes the program all the way to the end..

```shell
(gdb) n

Breakpoint 4, main () at src/main.rs:20
20      #[entry]
(gdb) n
(gdb) ^C
Program received signal SIGINT, Interrupt.
stim_practice::__cortex_m_rt_main () at src/main.rs:37
37              loop {
(gdb) 
```

JOE1994, 2020-07-05:
Update:
I tested the above example program on another board(`Nucleo-429ZI`), and it also exhibits the same behavior :crying_cat_face: 

So far I've tested two devices and they both exhibit the same behavior. Both belong to the `STM32F4` series.
* `stm32f407g-disc1`
* `Nucleo-F429ZI`

jonas-schievink, 2021-01-09:
Maybe try with probe-rs' ITM support?