rust-embedded/cortex-m #265: SWI / SVC support

thejpster, 2020-09-04:
Could we add intrinsics for calling Software Interrupts (the SWI or SVC instruction)? We probably also want a mechanism for marshalling some number of integer arguments into registers, and back out again on the other side of the SWI (i.e. in the kernel).

TockOS do this with `llvm_asm!`, as you can see at https://github.com/tock/tock/blob/198ff4173e85f6b8474893c6784d38c12ead9dbb/arch/cortex-m0/src/lib.rs#L161

jonas-schievink, 2020-09-04:
As described in https://github.com/rust-embedded/cortex-m/issues/58, this is not in general possible since the specific `svc` ABI depends on the environment

thejpster, 2020-09-04:
Ah. I guess it would be unreasonable to have an array of 256 function pointers to let you pick the SVC value...

jonas-schievink, 2021-01-09:
Once inline assembly is stable, we can provide a const-generic function like this:

```rust
fn svc<const NUM: u8>() {
    unsafe {
        asm!("svc {imm}", imm = const NUM);
    }
}
```

thejpster, 2021-01-16:
> Once inline assembly is stable, we can provide a const-generic function like this:
> 
> ```rust
> fn svc<const NUM: u8>() {
>     unsafe {
>         asm!("svc {imm}", imm = const NUM);
>     }
> }
> ```

Nice!

thejpster, 2021-02-10:
So reading around a bit on this, some syscall ABIs just set the SVC value to be zero, as it's expensive to pull it out on the receiving side (you have to read the instruction that caused the interrupt, which will be in a different cache...). So, could we at least have a function which lets you SVC 0, and optionally populate some registers with arguments.

jamesmunns, 2022-04-04:
Just tagging this issue as I've hit this as well, and we now have stable asm.

That being said, I'm probably also just going to go the "svc 0" route as well.

adamgreig, 2022-04-04:
I guess we'll want https://doc.rust-lang.org/unstable-book/language-features/asm-const.html for this one.

korken89, 2025-02-05:
Seems to be working now :)
https://rust.godbolt.org/z/fPK5Mh5sz