rust-embedded/cortex-m #267: inline-asm feature causes compile errors on cortex-m0 (samd21)

taurr, 2020-09-30:
Using version 0.6.3 with feature `inline-asm` enabled causes compile errors:

``` shell
error: invalid instruction, any one of the following would fix this:
  --> /home/johnny/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.3/src/asm.rs:39:13
   |
39 |             llvm_asm!("1:
   |             ^
   |
note: instantiated into assembly here
  --> <inline asm>:3:19
   |
3  |                   subs r1, $1
   |                   ^

note: instruction requires: thumb2
  --> /home/johnny/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.3/src/asm.rs:39:13
   |
39 |             llvm_asm!("1:
   |             ^
   |
note: instantiated into assembly here
  --> <inline asm>:3:19
   |
3  |                   subs r1, $1
   |                   ^

note: invalid operand for instruction
  --> /home/johnny/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.3/src/asm.rs:39:13
   |
39 |             llvm_asm!("1:
   |             ^
   |
note: instantiated into assembly here
  --> <inline asm>:3:28
   |
3  |                   subs r1, $1
   |                            ^

note: operand must be an immediate in the range [0,255]
  --> /home/johnny/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.3/src/asm.rs:39:13
   |
39 |             llvm_asm!("1:
   |             ^
   |
note: instantiated into assembly here
  --> <inline asm>:3:28
   |
3  |                   subs r1, $1
   |                            ^

note: operand must be a register in range [r0, r7]
  --> /home/johnny/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.3/src/asm.rs:39:13
   |
39 |             llvm_asm!("1:
   |             ^
   |
note: instantiated into assembly here
  --> <inline asm>:3:28
   |
3  |                   subs r1, $1
   |                            ^

error: aborting due to previous error

error: could not compile `atsamd-hal`
```

adamgreig, 2020-09-30:
It looks like this issue was about the cortex-m crate, so I've transferred it here. What Rust version are you using, and what target are you building for (in your `.cargo/config.toml` file or `--target` command line argument)?

taurr, 2020-09-30:
@adamgreig here's a little more info...

* rustc 1.48.0-nightly (8b4085359 2020-09-23)
* target: thumbv6m-none-eabi

Regarding the target, specifically I'm doing firmware for a [atsamd21g18](https://www.microchip.com/wwwproducts/en/ATSAMD21G18).

adamgreig, 2020-09-30:
Thanks for the info. I can reproduce the error on today's nightly-2020-09-29 and as far back as nightly-2020-08-28 but it's not present on nightly-2020-08-01, so this looks like a change or regression in nightly sometime in August. As a workaround you should be able to use 2020-08-01 (rustup install nightly-2020-08-01 --profile minimal; rustup target add thumbv6m-none-eabi --toolchain nightly-2020-08-01; cargo +nightly-2020-08-01 build).

@jonas-schievink, any ideas?

jonas-schievink, 2020-09-30:
Seems like a rustc regression, yeah. If you can bisect to a nightly we should be able to nail down the PR that caused it (was the LLVM upgrade in that range?).

That said, this doesn't seem like something that will be fixed, since `asm!` is now a thing.

adamgreig, 2020-09-30:
Ok, the last nightly that worked is nightly-2020-08-23 (rustc 1.47.0-nightly (663d2f5cd 2020-08-22)) and the first nightly that fails is nightly-2020-08-24 (rustc 1.47.0-nightly (5180f3da5 2020-08-23)).

jonas-schievink, 2020-09-30:
Commits in that range: https://github.com/rust-lang/rust/compare/663d2f5cd...5180f3da5

Looks like the LLVM 10 upgrade is indeed what caused this

Disasm, 2020-09-30:
It doesn't look like a rustc bug to me. One should be careful writing inlined asm code as it could be inlined into a code with different thumbness. You can't use some instructions both in ARM and Thumb mode when they do not exist in both modes.
I see two possible solutions:
* do not inline this delay code at all
* write code universal enough to support both modes

adamgreig, 2020-09-30:
But this is a thumbv6m-none-eabi target; it only runs thumb code and doesn't support ARM mode, so why would that be the issue here?

Disasm, 2020-09-30:
Hmm, you're right. Disregard my note then.

jonas-schievink, 2020-11-17:
This should be fixed in cortex-m 0.7

thalesfragoso, 2020-11-22:
And also on 0.6.4, not sure why this wasn't close automatically. Anyways, thanks for the report.

commit a624b6af (2020-10-26): Merge #274

274: Drop AT&T syntax from inline asm r=therealprof a=thalesfragoso

This was breaking compilation on thumbv6 targets and miscompiling on thumbv7.

Fixes #272 
Fixes #267
PR #274 (2020-10-26): Drop AT&T syntax from inline asm
This was breaking compilation on thumbv6 targets and miscompiling on thumbv7.

Fixes #272 
Fixes #267
closed 2020-11-22 by thalesfragoso