rust-embedded/cortex-m #271: Latest release of cortex-m-semihosting seems broken?

korken89, 2020-10-15:
After the 0.4.0 update of cortex-m-semihosting we are seeing breakage in RTIC CI pipeline as follows:

```console
error: failed to run custom build command for `cortex-m-semihosting v0.4.0`

Caused by:
  process didn't exit successfully: `/home/emifre/Git/rtic/cortex-m-rtic/target/debug/build/cortex-m-semihosting-6c5d0dd3173d1553/build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-semihosting-0.4.0/build.rs:15:14
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

No idea what is actually wrong here, but it seems something broke in the latest release?

jamesmunns, 2020-10-15:
Relevant build script code:

https://github.com/rust-embedded/cortex-m/blob/master/cortex-m-semihosting/build.rs#L10-L15

CC @jonas-schievink who I think worked on the prebuild static archives most recently in #262

jamesmunns, 2020-10-15:
We should probably mirror the build.rs changes made in this commit:

https://github.com/rust-embedded/cortex-m/commit/73a9d29b298100b4454e6ca17ca057a4dda0a91a#diff-d0d98998092552a1d3259338c2c71e118a5b8343dd4703c0c7f552ada7f9cb42

jonas-schievink, 2020-10-15:
Seems like `cargo package` doesn't include the right files here? We do have tests for cortex-m-semihosting and those worked fine prior to publishing.

What if we symlink `cortex-m-semihosting/bin` to `../bin`? Will `cargo package` include the files?

jonas-schievink, 2020-10-15:
Yanked 0.4.0

adamgreig, 2020-10-19:
@jonas-schievink, it seems to me like we could forward the `inline-asm` feature of c-m-sh to its c-m dependency and use the new `syscall` added to cortex-m, therefore removing c-m-sh's build script and prebuilt binary entirely?

jonas-schievink, 2020-10-19:
yeah that should work

adamgreig, 2020-10-19:
Though it will mean c-m-sh 0.4 requires c-m 0.7 (not yet released) which is unfortunate, hm.

adamgreig, 2020-10-22:
Well, that didn't work so well.

```
          rust-lld: error: duplicate symbol: __wfi
          >>> defined at inline.rs:13 (asm/inline.rs:13)
          >>>            bin/thumbv7em-none-eabihf.o:(__wfi) in archive /tmp/rustckWdISd/libcortex_m_semihosting-33bbc22c19d1c0ab.rlib
          >>> defined at asm.s:189
          >>>            cortex-m.o:(.text.__wfi+0x1) in archive /tmp/rustckWdISd/libcortex_m-c1c501809caa9026.rlib
```

nickray, 2020-11-28:
What's the status on this?

If I have a library that uses `cortex-m` 0.7, since 0.4.0 is yanked and 0.4.1 unpublished, I get a `cortex-m-semihosting` that forces `cortex-m` 0.6, which ends up with a `warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!` and an `error: failed to load bc of "cortex_m-faf0ca0469234fc4.cortex_m.2vla0p20-cgu.0.rcgu.o"`.

I'm a bit confused :)

adamgreig, 2020-12-02:
This should now be fixed by the new cortex-m-semihosting release v0.3.6, which supports either cortex-m 0.6 or cortex-m 0.7.

commit eefd6728 (2020-10-21): c-m-sh: fix missing prebuilt binaries. closes #271
commit 25f6613a (2020-10-22): Merge #273

273: c-m-sh: fix missing prebuilt binaries. closes #271 r=adamgreig a=adamgreig

Adding a symlink to `../bin` allows `cargo package` to include a copy of the binary files in the resulting tarball, which means the build script is able to link them as normal.

* We include the `-lto.a` files, but don't have an LTO_PLUGIN feature and so never use them
* While this should work for all platforms to use from crates.io, it's not clear what e.g. Windows will do with the symlink when trying to use this as a path/patch dependency directly from the git repository.

If it seems useful I could  [... cut]
closed 2020-12-02 by adamgreig