rust-embedded/cortex-m #295: Use `#[doc(cfg(...))]`

jonas-schievink, 2020-11-19:
A lot of our API is behind `#[cfg]`. Currently, that fact is only sometimes indicated in the documentation, which makes it hard to figure out, but as it turns out rustdoc has a feature for that: `#[doc(cfg(<...>))]`.

This can be used to decorate types and method with information about their availability. An example of this can be seen in the [`async-std`](https://docs.rs/async-std/1.7.0/async_std/) docs, which marks some APIs as `unstable`.

TDHolmes, 2021-12-15:
It looks like this is getting superseded by this automatically happening for us: https://github.com/rust-lang/rust/pull/89596

TDHolmes, 2021-12-15:
Once https://github.com/rust-lang/rust/issues/43781 lands, we should get this for free if I'm understanding things properly