rust-embedded/cortex-m #296: Cycle counter not available when debugger is not attached.

snakehand, 2020-11-23:
I wanted to count cycles on some code on a board where there was no debugger attached. I had to jump through several hoops to get there, and ended up with this monstrosity:

    let mut cp = cortex_m::peripheral::Peripherals::take().unwrap();
    unsafe {
        cp.DCB.demcr.modify(|r| r | (1<<24));
        fence(Ordering::SeqCst);
        // DWT::unlock();
        let ptr: *mut u32 = 0xE0000FB0 as *mut u32;
        *ptr = 0xC5AC_CE55;
        fence(Ordering::SeqCst);
        let ptr: *mut u32 = 0xE0001004 as *mut u32;
        *ptr = 0;
    }
    fence(Ordering::SeqCst);
    cp.DWT.enable_cycle_counter();

Several problems had to be adressed when applying the solution from https://stackoverflow.com/questions/36378280/stm32-how-to-enable-dwt-cycle-counter.

1) The compiler could re-order the writes, so I had to place memory barrier liberally around.
2) DEMCR tracing needed to be enabled
3) The unlock command wrote to the wrong register, 0xE0000FB0 has ITM as the base, and not DWT ( I think it ended up at 0xE0001FB0 instead )
4) The SO article states that it is nececary to write to the cycle count register first. There is no safe facility for doing so

samueltardieu, 2020-11-25:
I noticed your issue. While I'm not a developer of this crate, I will address some remarks. I'll number them the same as you did.

1. If you use volatile write, the compiler cannot reorder them and you don't need barriers.
2. `cp.DCB.enable_trace()` can do that.
3. There are several registers named LAR on Cortex-M7, including one in DWT. If you want to unlock the DWT you have to use this one. It is the DWT you need to unlock in order to be able to enable to write into the control register to enable the counter.
4. I don't think you need to write to the cycle count registers, and they do that to reset it to a known value (0), which is its reset value anyway.

You can find the whole sequence [in this RTIC book excerpt](https://rtic.rs/0.5/book/en/by-example/timer-queue.html):

```rust
cp.DCB.enable_trace();
DWT::unlock();
cp.DWT.enable_cycle_counter();
```

What do you propose to change in the `cortex-m` crate? Is there anything else you would need to activate the cycle counter?

snakehand, 2020-11-26:
I think you have answered 1,2,4 - so I knly have questions regarding 3:
`    cp.DCB.enable_trace();
    DWT::unlock();
`

compiles to:

` 8000696:       f64e 50fc       movw    r0, #60924      ; 0xedfc
 800069a:       f04f 0800       mov.w   r8, #0
 800069e:       f2ce 0000       movt    r0, #57344      ; 0xe000
 80006a2:       6801            ldr     r1, [r0, #0]
 80006a4:       f041 7180       orr.w   r1, r1, #16777216       ; 0x1000000
 80006a8:       6001            str     r1, [r0, #0]
 80006aa:       f241 0104       movw    r1, #4100       ; 0x1004
 80006ae:       f64c 6055       movw    r0, #52821      ; 0xce55
 80006b2:       f2ce 0100       movt    r1, #57344      ; 0xe000
 80006b6:       f2cc 50ac       movt    r0, #50604      ; 0xc5ac
 80006ba:       f8c1 8000       str.w   r8, [r1]
 80006be:       f8c1 0fac       str.w   r0, [r1, #4012] ; 0xfac
`
 So the unlock write happens to address 0xe0001fb0  which I think is incorrect. This might have something to do with fact that this run was on Cortex-M4, which does not require unlock. I will move to Cortex-M7 later where unlock is required, and see if the behaviour is correct there. If this becomes an issue, I can re-open the case.

closed 2020-11-26 by snakehand