rust-embedded/cortex-m #301: Version 0.3.6 is not semver-compatible

samueltardieu, 2020-12-02:
The prototype of `hprintln!()` has changed between 0.3.5 and 0.3.6, version should be bumped to "0.4.x".

I had the experience today. Some of my students had imported a few days ago a library skeleton containing a dependency over "0.3.3" or any compatible version, and the code uses `hprintln!().unwrap()`. Everything worked fine and it created a `Cargo.lock` depending on 0.3.3 for their application.

Some other students imported the same code today, and since they didn't have a `Cargo.lock` yet one got created with the highest compatible version, i.e. "0.3.6". Unfortunately, the code containing `hprintln!().unwrap()` now doesn't compile anymore.

adamgreig, 2020-12-02:
Ah, sorry! I didn't realise there were any breaking changes still on master, but it looks like https://github.com/rust-embedded/cortex-m-semihosting/pull/48 was included and should not have been. I've yanked 0.3.6 and will prepare a compatible 0.3.7 instead.

adamgreig, 2020-12-02:
This should be fixed by 0.3.7 which is now released. Sorry for the disruption!

closed 2020-12-02 by adamgreig

samueltardieu, 2020-12-02:
Thanks @adamgreig for your very fast reaction!