rust-embedded/cortex-m #309: Unable to debug application when using library version 0.7.0

wbuck, 2020-12-11:
First, I'am able to successfully flash the `STM32F303x` and run the application.
The issue I'm facing is attempting to debug the application with `GDB`/`OpenOCD`.

When issuing a `cargo run` I see the following messages (emphasis mine):

>Reading symbols from target/thumbv7em-none-eabihf/debug/app...
**D*W_FORM_strp pointing outside of .debug_str section*** [in module /Users/Me/Projects/embedded_template/target/thumbv7em-none-eabihf/debug/app]

>(No debugging symbols found in target/thumbv7em-none-eabihf/debug/app)

As a result I'am unable to set break points or use `layout src`.
This only occurs when using `cortex-m` version `0.7.0`.

If I use version `0.6.4` I'am able to successfully debug the application.

Aside from the `memory.x` file is there additional configuration needed for the newest version of the library?
Are specific versions of the library only compatible with specific `MCU`'s? 
Does the version of the `cortex-m` library require a specific version of `GDB`?

Below is my `memory.x` file if it helps:

    MEMORY
    {
        CCRAM : ORIGIN = 0x10000000, LENGTH = 8K
        FLASH : ORIGIN = 0x08000000, LENGTH = 256K
        RAM : ORIGIN = 0x20000000, LENGTH = 40K
    }
    _stack_start = ORIGIN(CCRAM) + LENGTH(CCRAM);

jonas-schievink, 2021-01-19:
This seems to be an upstream issue, it is fixed on nightly Rust versions.

wbuck, 2021-01-19:
OK great, thanks @jonas-schievink

closed 2021-01-25 by bors[bot]

wbuck, 2021-01-25:
@jonas-schievink so to be clear if I pass `-Clinker-plugin-lto` to the compiler I should be able to use version `0.7`?

adamgreig, 2021-01-25:
This is now fixed by the new 0.7.1 release (containing #318).

adamgreig, 2021-01-25:
> @jonas-schievink so to be clear if I pass `-Clinker-plugin-lto` to the compiler I should be able to use version `0.7`? Or should I wait for version `0.7.1` to be released?

Hopefully 0.7.1 solves the issue! Thanks for reporting the bug, sorry we couldn't get it fixed sooner.

commit 4c624da7 (2021-01-25): Use LTO instead of removing the panic handler manually. Closes #309
commit 2082aca4 (2021-01-25): Merge #318

318: Use LTO instead of removing the panic handler manually. Closes #309 r=jonas-schievink a=adamgreig