rust-embedded/cortex-m #314: DW_FORM_strp pointing outside of .debug_str section

pozix604, 2021-01-13:
Crate version 0.7.0 results in GDB error message `DW_FORM_strp pointing outside of .debug_str section` while crate 0.6.0 doesn't. Here is how to reproduce, using the example project referenced in the Embedded Rust book:

```
cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart
```

In `Cargo.toml`, replace the `cortex-m` dependency line:

```
cortex-m = "0.7.0"
```

Build:

```
cargo build
```

Load binary with debugger:

```
$ arm-none-eabi-gdb -q target/thumbv7m-none-eabi/debug/app

Reading symbols from target/thumbv7m-none-eabi/debug/app...
DW_FORM_strp pointing outside of .debug_str section [in module /Users/xxx/scm/cortex-m-quickstart/target/thumbv7m-none-eabi/debug/app]
(No debugging symbols found in target/thumbv7m-none-eabi/debug/app)
```

As a result, debugging symbols are not available. This does not happen with 0.6.0.

jonas-schievink, 2021-01-13:
Looks like a rustc or LLD bug

jonas-schievink, 2021-01-13:
Duplicate of https://github.com/rust-embedded/cortex-m/issues/309

closed 2021-01-13 by jonas-schievink