rust-embedded/cortex-m #329: macOS produces different archives from Linux

jonas-schievink, 2021-02-09:
Seen in https://github.com/rust-embedded/cortex-m/pull/328, this means that running `cargo xtask assemble` on macOS will produce different archives from what CI expects.

Not sure what the cause of this is, or if this is even fixable, but we should make sure that users on macOS can contribute to this crate without problems.

therealprof, 2021-02-09:
Hm, do we know what the differences are? Let me try that. 😅

therealprof, 2021-02-09:
The generated code is 100% identical but the order of the symbols in the generated objects is not stable. Interestlingy that's only the case for the non`-lto` archives, the `-lto` archives are fine.

therealprof, 2021-02-09:
I tried a ton of different options and a few different nightly versions but I couldn't get the symbol order to line up. 🤷🏻‍♂️

jonas-schievink, 2021-02-09:
Perhaps we should relax the check in CI then

therealprof, 2021-02-09:
Hm, we could change the check to only verify the `-lto` ones, should be good enough to detect subtle code changes?

newAM, 2024-06-30:
I'm triaging Cortex-M issues today.

I tried to do the basic CI tasks on an M1 Mac and everything works now as far as I can tell, and we no longer have pre-compiled blob.s

I'm going to close this issue, please re-open if I missed anything!

closed 2024-06-30 by newAM