rust-embedded/cortex-m #332: SCB::vect_active() should be 9? bits

Dirbaio, 2021-02-14:
Currently `vect_active()` returns the irq number as a u8, but the SCB.ICSR.VECTACTIVE field is 9 (or more?) bits. This means that the interrupt numbers will wrap on chips with a lot of interrupts.

adamgreig, 2021-02-15:
Yep, 9 bits. On cortex-m4 and -m7 the maximum number of interrupts is 240, which I expect is where this bug originally came from, but on -m33 it's allowed to be up to 480, which would definitely cause issues. This is basically what motivated removing bare_metal::Nr in favour of traits per architecture platform, and cortex_m::InterruptNumber being a u16... unfortunately this got missed, and since it's a breaking change it will have to wait for 0.8 I guess.

jamesmunns, 2021-02-26:
Could we add a new method that takes a U16 without removing the u8 one? That way we can support new targets, and just switch over the U16 one to be an alias for the future 8->16 upgraded method?

commit a2d8f6de (2021-12-31): `SCB.ICSR.VECTACTIVE` is 9 bits, not 8

Closes #332
commit bef6ed18 (2021-12-31): Merge #373

373: `SCB.ICSR.VECTACTIVE` is 9 bits, not 8 r=adamgreig a=TDHolmes

Closes #332
PR #373 (2021-12-31): `SCB.ICSR.VECTACTIVE` is 9 bits, not 8
Closes #332
closed 2021-12-31 by bors[bot]