rust-embedded/cortex-m #333: v0.6: Some peripherals are not re-exports from v0.7

mattico, 2021-02-16:
Half of them are re-exports: CPUID, DCB, DWT, FPB, FPU, SYST, TPIU
Half of them are not: CBP, ITM, MPU, NVIC, SCB

I'm guessing this was intentional, but I'm wondering what the motivation was. It doesn't look like there were substantial changes to those types in v0.7. Not re-exporting those types can cause issues (https://github.com/stm32-rs/stm32h7xx-hal/pull/185/commits/7d875d69bb3cbfdc3cba4bc59a3141b9845f4c2f).

adamgreig, 2021-02-16:
There were at least _some_ breaking changes to those peripherals, so we couldn't re-export them in v0.6 without breaking existing v0.6 users. The ones that were re-exported from v0.7 have to be _exactly_ the same types to not break backwards compatibility.

The downside is indeed that you can't give e.g. a library using 0.6 a changed peripheral from 0.7's Peripherals struct. As far as I can tell, the alternative is not re-exporting _any_ of the types, and only sharing the `TAKEN` flag to allow linking multiple versions together. If you have any ideas on how we could do better, I'm all ears, though, this is a regular pain point every upgrade because of how coupled the PACs and HALs are to cortex-m versions.

mattico, 2021-02-16:
Understandable, just wanted to make sure!

closed 2021-02-16 by mattico