rust-embedded/cortex-m #336: cannot find type `RegisterBlock` in module `self::mpu` on i686-pc-windows-gnu

dimpolo, 2021-03-19:
My default toolchain is  i686-pc-windows-gnu.
I updated the cortex-m dependency today and that caused CI to break.

Presumably introduced in https://github.com/rust-embedded/cortex-m/commit/1c198a1f53b0fdef53c723843331a059da6cb6a3

Could this line be changed to `#[cfg(not(armv8m))]` ?
https://github.com/rust-embedded/cortex-m/blob/1c198a1f53b0fdef53c723843331a059da6cb6a3/src/peripheral/mpu.rs#L6

adamgreig, 2021-03-19:
Yes, I think that change could be made without even a breaking change, sounds OK to me. Would you like to open a PR? Also, is there any particular reason to build cortex-m on i686? Usually in CI we'll build it for the thumb target; the only reason anything builds for x86 is because docs.rs used to require it (but these days can be told to build for the thumb targets too, so...).

dimpolo, 2021-03-19:
Thanks for the quick response
Sure, I'll write a PR :)

The reason might be my misunderstanding of how to write `no_std` unit tests haha.

If I run `cargo test` I get `can't find crate for test`
So I added `#![cfg_attr(not(test), no_std)]` and got `can't find crate for std`
Which is why I ended up with `cargo test --target i686-pc-windows-gnu`
And the reason I use `i686-pc-windows-gnu` is the debugger in CLion

closed 2021-03-19 by bors[bot]